zephyrproject-rtos/zephyr #40163: Use correct clock frequency for systick+DWT

Michel-Haber, 2021-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
 zephyr/arch/arm/core/aarch32/cortex_m/timing.c uses CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC to determine the Arm core's frequency, but this variable doesn't necessarily have the right value if CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME is set.

**Describe the solution you'd like**
sys_clock_hw_cycles_per_sec() should be used to replace the config constant when CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME is set

**Additional context**
I believe this is a latent error from before frequency change at runtime was introduced.

carlescufi, 2021-11-09:
@Michel-Haber thanks for this, would you mind sending a Pull Request to fix it?

Michel-Haber, 2021-11-09:
Sure, @carlescufi PR #40219

PR #40219 (2021-11-16): timing: use runtime cycles for cortex-m systick
Use sys_clock_hw_cycles_per_sec() instead of
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC to determine clock cycles.

Fixes #40163
closed 2021-11-16 by carlescufi