zephyrproject-rtos/zephyr #40166: A cmake re-run after a git commit will cause all source files to re-compile.

tejlmand, 2021-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
Don't rebuild everything when git revision changes.

Today, if `git revision` changes and CMake re-runs then all source files are rebuilt, even if there are no changes.

To reproduce:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/hello_world/
Including boilerplate (Zephyr repository): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
...
-- Build files have been written to: /projects/github/ncs/zephyr/build
$ ninja -Cbuild
...
[150/150] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%

# Create new commit and re-run CMake:
$ touch new_file.txt; git add new_file.txt; git commit -s -m"new file added"
$ cmake -Bbuild samples/hello_world/
Including boilerplate (Zephyr base (cached)): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
....

$ ninja -Cbuild
...
[145/145] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%
```

As can be seen, all source files are being re-compiled even if they have no changes.

This happens because `BUILD_VERSION` is added as a compile definition.

**Describe the solution you'd like**

The BUILD_VERSION should be place in a header file, for example `version.h` so that only files including this header is rebuilt when the header is updated.

For example like this:
```
$ ninja -Cbuild
...
[150/150] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%

# Let's touch `version.h` and rebuild:
$ touch build/zephyr/include/generated/version.h
$ ninja -Cbuild
...
[10/10] Linking C executable zephyr/zephyr.elf
```

as can see, in last case, only source files depending on `version.h` like `version.c` and `banner.c` are being re-built.

erwango, 2021-11-09:
@tejlmand Seems like you duplicated this issue, cf https://github.com/zephyrproject-rtos/zephyr/issues/40167

tejlmand, 2021-11-09:
Had some issues with networking / github yesterday.
So most likely that's the cause for the double post.

Thanks for noticing.

Closed.

closed 2021-11-09 by tejlmand