zephyrproject-rtos/zephyr #40167: A cmake re-run after a git commit will cause all source files to re-compile.

tejlmand, 2021-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
Don't rebuild everything when git revision changes.

Today, if `git revision` changes and CMake re-runs then all source files are rebuilt, even if there are no changes.

To reproduce:
```
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -Bbuild samples/hello_world/
Including boilerplate (Zephyr repository): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
...
-- Build files have been written to: /projects/github/ncs/zephyr/build
$ ninja -Cbuild
...
[150/150] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%

# Create new commit and re-run CMake:
$ touch new_file.txt; git add new_file.txt; git commit -s -m"new file added"
$ cmake -Bbuild samples/hello_world/
Including boilerplate (Zephyr base (cached)): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
....

$ ninja -Cbuild
...
[145/145] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%
```

As can be seen, all source files are being re-compiled even if they have no changes.

This happens because `BUILD_VERSION` is added as a compile definition.

**Describe the solution you'd like**

The BUILD_VERSION should be place in a header file, for example `version.h` so that only files including this header is rebuilt when the header is updated.

For example like this:
```
$ ninja -Cbuild
...
[150/150] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       16864 B         1 MB      1.61%
            SRAM:        5440 B       256 KB      2.08%
        IDT_LIST:          0 GB         2 KB      0.00%

# Let's touch `version.h` and rebuild:
$ touch build/zephyr/include/generated/version.h
$ ninja -Cbuild
...
[10/10] Linking C executable zephyr/zephyr.elf
```

as can see, in last case, only source files depending on `version.h` like `version.c` and `banner.c` are being re-built.

SebastianBoe, 2021-11-08:
Nice solution :)

marc-hb, 2021-11-10:
Of course it will be even better then `version.h` is checksummed and its timestamp put back in time to fool ninja (cause https://github.com/ninja-build/ninja/issues/1459) but in the mean time this would already be a huge leap forward! +1

FWIW here's SOF's `version.h` https://github.com/thesofproject/sof/blob/main/scripts/cmake/version.cmake. Not the best CMake code ever but there could be some re-usable bits there, who knows?

PR #42527 (2022-02-08): cmake: version.h generation performed at build time
Fixes: #39503
Fixes: #40167

This commit moves `BUILD_VERSION` CMake variable from a compile
definition to be a define inside version.h.

Besides the benefit of having related settings grouped in a common
header, it also means that an updated `BUILD_VERSION` value does not
need to trigger re-compilation of all source files.

When using compile definitions, CMake cannot tell whether a given source
files uses the definition or not, and hence all sources must be
recompiled to be sure they are up-to-date.

Placing `BUILD_VERSION` in version.h, th [... cut]
commit 91709778 (2022-02-08): cmake: version.h generation performed at build time

Fixes: #39503
Fixes: #40167

This commit moves `BUILD_VERSION` CMake variable from a compile
definition to be a define inside version.h.

Besides the benefit of having related settings grouped in a common
header, it also means that an updated `BUILD_VERSION` value does not
need to trigger re-compilation of all source files.

When using compile definitions, CMake cannot tell whether a given source
files uses the definition or not, and hence all sources must be
recompiled to be sure they are up-to-date.

Placing `BUILD_VERSION` in version.h, t [... cut]
closed 2022-02-08 by nashif