zephyrproject-rtos/zephyr #40189: k_poll infrastructure can miss "signals" in a heavily contended SMP system

venkatesh-eq, 2021-11-09:
I was testing he k_poll mechanism in Zephyr and found that it can miss events from the producer in SMP.

**Steps to reproduce:**
1) Create two threads - producer and consumer threads. have an event to of type K_POLL_TYPE_SIGNAL & K_POLL_MODE_NOTIFY_ONLY in consumer thread. 
2) Producer thread will call k_poll_signal_raise() to wakeup the consumer after producing something in a shared memory (like a ring buffer interface).
3) Consumer thread will call a handler when event is signalled and run it to completion/process the ring to empty and goes back again to k_poll to pend on event. Before going back to pend, the consumer thread has to clear the "signalled" field to zero.

Now depending how 2) and 3) are ordered - step 3) could set "signalled" back to zero and lose the update from the producer thread in step 2).

Shouldn't we take a snapshot of the events/signals when exiting k_poll in the critical section and pass that to the destination thread? Once, a thread calls k_poll again then "OR" the pending events with events that the thread has not handled so far (if any)? We should not clear "signalled" field of the same signal object without any protection.

andyross, 2021-11-09:
Do you have code that exhibits this behavior?  I'm looking at the code and it looks OK to me.  The signaled state is cleared before the thread returns out of k_poll(), subsequent signals at that point will just set it high again and the next caller into k_poll() will return synchronously.

venkatesh-eq, 2021-11-09:
The signalled state is cleared by the consumer thread outside k_poll. Extract from documentation:

struct k_poll_signal signal;
void do_stuff(void)
{
    k_poll_signal_init(&signal);

    struct k_poll_event events[1] = {
        K_POLL_EVENT_INITIALIZER(K_POLL_TYPE_SIGNAL,
                                 K_POLL_MODE_NOTIFY_ONLY,
                                 &signal),
    };

    for (;;) {
        k_poll(events, 1, K_FOREVER);

        if (events[0].signal->result == 0x1337) {
            // A-OK!
        } else {
            // weird error
        }

        events[0].signal->signaled = 0;
        events[0].state = K_POLL_STATE_NOT_READY;
    }
}

andyross, 2021-11-09:
OK... I think maybe I'm starting to see what you're trying to say.  I thought you were saying that a signal could be missed and leave a correctly-suspended thread waiting forever.  But what you're actually seeing is that the signal arrives and races with application code doing the clear of the "signaled" field, so the app thread (which woke up correctly, at the right time) fails to notice that the signal was flagged, because its "signaled" field is already clear.  So then it goes to sleep again and is suspended (correctly) awaiting the next call to signal_raise().

That's a lot of words, but does that match your report?  If not, can you submit an exerciser showing the hang that I can try to dig into?

So, that said, how is that not an app bug?  The glitch here is that you have a call to k_poll_signal_raise() and an assignment to "event.signal->signaled" that are unsynchronized.  Both of those are happening in application code, there's nothing the kernel can do.

I totally agree, by the way, that:

1. The "signal" variant of k_poll event is naturally racy like this, and sort of a booby trap.   The other IPC mechanisms provide their own synchronization and avoid this, but signal requires that the app do it.
2. The documentation is silent on this requirement, and in fact that sample code you found is sorta borderline irresponsible not to show any locking at all.

Maybe we should fix this as a doc thing?

andyross, 2021-11-09:
Stating the same truth in a different way:  it's not correct for a thread waking up from k_poll() to **read** the value of event.signal->signaled (to "check if the signal was raised").  The only correct way to use that field is for the polling thread to set it to zero immediately before calling k_poll().  If you do that, you are guaranteed to be woken up (or already awake) immediately after the call to k_poll_signal_raise().

If you need more behavior (i.e. "was the signal raised once or twice?"), you need to implement that externally via application synchronization.

venkatesh-eq, 2021-11-10:
> ..you're actually seeing is that the signal arrives and races with application code doing the clear of the "signaled" field, so the app thread (which woke up correctly, at the right time) fails to [...]

This is the issue. you gotit right.

Are you asking to rely on the fact that we woke up from k_poll and simple call all the event list handlers (which are of type SIGNAL)? Not look into the k_poll_event->state == K_POLL_STATE_SIGNALED ?

The better thing would be for the k_poll to return a snapshot of all events set (like a bitmap) signal->signalled is not touched by app?

andyross, 2021-11-10:
I'm saying don't look into the signal state without proper synchronization.  The fields in `struct k_poll_signal` might look like they're kernel-managed, but in fact they're only assigned synchronously under the command of the application (e.g. in k_poll_signal_raise(), k_poll_signal_init(), etc...).  You need to make sure those moments are properly understood.  Other IPC mechanisms do this for you, signal doesn't.

But I agree that the signal thing is a little half-baked and that a better API should be possible.

And I **definitely** agree that our documentation is broken for failing to note this behavior and call it out really clearly.

Also, for your specific use case, it really sounds like you should be polling on semaphores instead.

venkatesh-eq, 2021-11-11:
Ok. please fix documentation and i will look at using k_poll using semaphores instead? BTW are k_sem_give and k_poll_signal_raise() legal to be called from ISRs?

andyross, 2021-11-11:
Yes, both are interrupt-safe.  With only a few exceptions, everything in Zephyr can be called in an ISR, the only thing you can't do is block (obviously).  But blocking calls almost all take timeout arguments which can accept K_NO_WAIT for interrupts context.

PR #41987 (2022-01-20): doc: Clarify k_poll() behavior with respect to signal synchronization
Bug #40189 tripped over an interesting synchronization scenario that
wasn't called out in the docs.  Poll signals are level triggered, and
if you're adjusting the level from a racing context (e.g. resetting it
before the next "event" from an ISR or another thread) the polling
thread might wake up but then miss the event.  Mention this case
explicitly in documentation.

Fixes #40189
commit aa825d77 (2022-01-20): doc: Clarify k_poll() behavior with respect to signal synchronization

Bug #40189 tripped over an interesting synchronization scenario that
wasn't called out in the docs.  Poll signals are level triggered, and
if you're adjusting the level from a racing context (e.g. resetting it
before the next "event" from an ISR or another thread) the polling
thread might wake up but then miss the event.  Mention this case
explicitly in documentation.

Fixes #40189
closed 2022-01-20 by nashif