zephyrproject-rtos/zephyr #40198: Shell module doesn't work on main branch for esp32 board

andrei-edward-popa, 2021-11-09:
**Describe the bug**
On the main branch, I think something is crashed on the shell part because when I load the code into the board nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b esp32 samples/subsys/shell/shell_module
2. west flash

**Logs and console output**
Nothing is output to the console.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Toolchain ESP-IDF
 - On the main branch

**Additional context**
Can you check this?

andrei-edward-popa, 2021-11-09:
@sylvioalves I think something is broken in uart driver (drivers/serial/uart_esp32.c) because with previous version it works.

sylvioalves, 2021-11-09:
Sure, that uart update should be pre-kernel initialization and not post kernel. Submitting a PR.  
Thanks for pointing that out.

andrei-edward-popa, 2021-11-09:
@sylvioalves I tried to put PRE_KERNEL_1 instead of POST_KERNEL in driver, but still doesn't work

sylvioalves, 2021-11-09:
You need to use PRE_KERNEL_2 because pinmux needs to be initilialized before.

PR #40203 (2021-11-09): drivers: uart: esp32: set device to initialize pre-kernel
Uart post kernel initialization does not allow starting shell
properly. As UART driver depends on pinmux, uart requires
to be initialized as PRE_KERNEL_2.

Closes #40198
closed 2021-11-09 by cfriedt