zephyrproject-rtos/zephyr #40201: runners:jlink: fail to flash board with protected sector when erase requested

Zzzwierzak, 2021-11-09:
**Describe the bug**
After change #39734 it is not possible to flash board with protected memory area. When erase of board is requested in jlink runner args, protection of memory cause error during mass erase and flash procedure stops. Issue is visible on stm32wb55 when part of memory that is used by second core is protected against erase. 

**To Reproduce**
Try to flash board with part of memory configured as protected with erase option enabled for jlink runner. 
board_runner_args(jlink "--device=STM32WB55xx" "--speed=4000" "--reset-after-load" "--erase")

**Expected behavior**
Board flashed successfully.

**Environment (please complete the following information):**)
 - Zephyr SDK 13.1
 - Zephyr: 404f67ab5e571b5e209ac59adc99edc15a3475d9

**Log**
J-Link>connect
Device "STM32WB55XX" selected.

Connecting to target via SWD
Found SW-DP with ID 0x6BA02477
DPIDR: 0x6BA02477
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: AHB-AP (IDR: 0x84770001)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
ROMTbl[0][6]: E0043000, CID: B105900D, PID: 005BB906 CTI
SetupTarget() start
WARNING: Secure Firmware detected @ 0x080CA000
Connecting to target AP[0]/Cortex-M4
CORE ID AP[0]:0x410FC241
CORE ID AP[0]/Cortex-M4 halted!
SetupTarget() end
Cortex-M4 identified.
J-Link>erase
Erasing device...
Comparing flash   [100%] Done.
Erasing flash     [100%] Done.

****** Error: Failed to erase chip @ address 0x08000000 (sector is locked)
Failed to erase chip.
Failed to execute RAMCode for chip erase!
J-Link: Flash download: Total time needed: 4.494s (Prepare: 0.025s, Compare: 0.000s, Erase: 4.461s, Program: 0.000s, Verify: 0.000s, Restore: 0.007s)
ERROR: Erase returned with error code -5.
J-Link>

carlescufi, 2021-11-09:
@bbilas @JordanYates FYI

bbilas, 2021-11-09:
Please revert this commit and put the log without this change + return code of the flash command. Basically that change shouldn't have an impact on that.

cfriedt, 2021-11-09:
@bbilas - can you send a PR to fix?

bbilas, 2021-11-10:
@cfriedt I don't have such hardware to reproduce that error so I'll try to use MIMXRT1060-EVK with JLink BASE instead. Any information related to it would be appreciated.

erwango, 2022-01-11:
@bbilas Did you had a chance to make some progress on this ?

bbilas, 2022-01-11:
No, I didn't. It seems that I need that HW to be able to reproduce that issue but I'm lacking that.

Zzzwierzak, 2022-01-25:
@bbilas, @erwango  I've checked on 25e6803705b25602eb0b730a03a254c354bfae9e and issue still exist.

erwango, 2022-01-25:
@Zzzwierzak Would you mind a try to provide what was suggested in https://github.com/zephyrproject-rtos/zephyr/issues/40201#issuecomment-964219844 ?

erwango, 2022-01-25:
My 2 cents on this issue:
- There's a protected fw (BLE f/w used on M0) on stm32wb socs (cf trace `WARNING: Secure Firmware detected @ 0x080CA000`)
- The jlink command request a full chip erase from what I can understand(` "--erase"`) which is not possible due to the presence of protected sectors and hence the command fails
- Issue wasn't visible before https://github.com/zephyrproject-rtos/zephyr/pull/39734, since the jlink command was continued on the already present erase command error

Based on this, I don't think that present behavior is an actual issue.
There are 2 options:
- Don't use --erase command at all
- If available, use a sector erase command if possible (or alike). This is what I'm using for this board on our bench with STM32CubeProgrammer tool (`STM32_Programmer_CLI -c port=SWD  freq=4000 ap=0 mode=UR  -e [0 255]'` > erasing sectors 0 to 255)

Zzzwierzak, 2022-01-25:
@erwango Output from Jlink without change is same( Error -5), but error from erase is discarded silently, after change it is breaking west flash in CI, that cause test fail with flash error. 
Unfortunately testing without erase is not stable, and using STM32CubeProgrammer is not possible.

erwango, 2022-01-25:
> @erwango Output from Jlink without change is same( Error -5), but error from erase is discarded silently, after change it is breaking west flash in CI, that cause test fail with flash error.

I understand the problem on your side, but this doesn't sounds like an actual issue as returning an error on erase error is a sound behavior. Support for your use case would likely require an enhancement :
- add support for sector erase
- add option not to fail on erase issue

Zzzwierzak, 2022-01-25:
@erwango Both options are good for me.

bbilas, 2022-01-25:
> I understand the problem on your side, but this doesn't sounds like an actual issue as returning an error on erase error is a sound behavior. Support for your use case would likely require an [...]

Support for sector erasing is the most suitable solution in my opinion only if it's supported by jlink.

bbilas, 2022-01-25:
@Zzzwierzak Would you mind testing this https://github.com/zephyrproject-rtos/zephyr/pull/42144 MR? It introduces erasing specified flash range but due to lack of hardware, I'm not able to test it. 

Example:
`west flash -d build --erase-range 0x60000000 0x607FFFFF`

Zzzwierzak, 2022-01-31:
@bbilas Yes, it's working with stm32wb55.

bbilas, 2022-02-06:
I guess we can close it?

Zzzwierzak, 2022-02-07:
@bbilas Yes.

bbilas, 2022-02-07:
@Zzzwierzak I don't have any permission to manage this issue, so could you do that as the author?

closed 2022-02-08 by Zzzwierzak