zephyrproject-rtos/zephyr #40202: [BT controller] Periodic advertising synchronization not re-established after advertiser reset without scan disable

ppryga, 2021-11-09:
**Describe the bug**
When advertiser device is restarted and gets new MAC address it is possible that periodic advertising sync won't be re-established.
The issue may occur when duplicate advertising filtering is enabled.
If scanner receives first advertisement with new advertiser MAC address before periodic advertising timeouts, then the address  will be available in duplicate filter list. 
When periodic advertising is lost and scanning does not go disable/enable cycle, then new sync will not be created because the advertiser will be filtered out.

**To Reproduce**
Steps to reproduce the behavior:
1. Build direction_finding_connectionless_tx sample as an advertiser.
2. In direction_finding_connectionless_rx comment out scan_disable() in main function.
3. Build direction_finding_connectionless_rx sample as a scanned.
4. Run both.
5. Reset advertiser once and see there is no sync re-establised.
6. Reset advertiser again and see new sync.

**Expected behavior**
Synchronization should be re-established every time advertiser is reset.

I'm not sure if the issue may be solved by something else than clearing the whole duplicate filtering list or at least removing all entries that point to periodic advertising.

**Impact**
Moderate. There are workarounds:
- go through scanner disable/enable cycle.
- disable duplicates filtering.

cvinayak, 2021-12-03:
I do not see a solution in the Controller, as the Controller does report any new advertiser when scanning is active with duplicate filtering. But if the application looks for new advertising reports after sync lost event, it will not get it as all devices discovered before the sync lost event are filtered out.

The correct application behavior shall be to cycle scan enable if new advertising reports are needed. (The direction_finding_connectionless_tx sample does cycle scan enable, and disabling the code is not advisable).

This issue may be closed, as there is no change required in controller implementation.

carlescufi, 2021-12-16:
@ppryga please weigh in and close if appropriate.

ppryga, 2021-12-16:
Ok, I'm fine with that. Closing the issue.

closed 2021-12-16 by ppryga