zephyrproject-rtos/zephyr #40208: BBC micro:bit v2 display driver

carlescufi, 2021-11-09:
This issue is here to track discussion and resolution of the current situation with the BBC micro:bit v2 display driver.

There have been 3 attempts so far at it:

- Extending the v1 driver: https://github.com/zephyrproject-rtos/zephyr/pull/32532
- New, native nrfx-based: https://github.com/zephyrproject-rtos/zephyr/pull/36280
- Extending the v1 driver, again: https://github.com/zephyrproject-rtos/zephyr/pull/40144

Copying the relevant people here:
@LingaoM @peterniebert @chudov

carlescufi, 2021-11-16:
Approach suggested after a chat with @gmarull and @jfischer-no:

1. Implement a native driver, similar to the [CODAL](https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/NRF52LedMatrix.cpp) one, and make it expose the display API as in `include/drivers/display.h`.
2. Extend the character framebuffer API in `include/display/cfb.h` so that it supports the feature set of the current v1 driver. Namely that includes scrolling and a few tidbits more.
3. Have applications use a combination of the `cfb` and `display` APIs to drive the display

carlescufi, 2021-11-17:
@peterniebert in [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/40144#issuecomment-962395347) you wrote:

>However, the whole thing remains terribly limited and inefficient, in particular compared to the driver present in CodeMaker software (based on NRF5SDK), for instance, that driver allows for changing brightness.
Moreover, the original author of the Zephyr driver tried to be smart and use line multiplexing (showing several pixels in the same row or same column). If you think about it, that is a cause of trouble in one of two ways: Either several LEDs take their current through the same resistance, but then, the more leds you switch on, the less bright they will be, leading to an inhomogenous brightness. Or the current of all LEDs adds up and passes through a single GPIO pin exceeding the specs of that pin, and this might over time damage the pin.

But in fact, the [official driver](https://github.com/lancaster-university/codal-microbit-v2/blob/master/source/NRF52LedMatrix.cpp) refreshes the screen by rows, although it does, as you mentioned, provide brightness control (generating a PWM signal with GPIOTE and PPI) and light sensing.

chudov, 2021-12-08:
And in #40968 it was done how it should be :)

carlescufi, 2021-12-21:
The micro:bit now drives its display using a new, unified nRF LED matrix display, compatible with v1 and v2.

closed 2021-12-21 by carlescufi

zpm1066, 2021-12-21:
Thank you all.

carlescufi, 2021-12-22:
@LingaoM @peterniebert @chudov this work is now complete, feel free to test the current v1 and v2 compatible driver on `main`.

chudov, 2021-12-23:
@carlescufi thank you!