zephyrproject-rtos/zephyr #40215: RSSI in periodic adv. callbacks always -127 (sync_recv and cte_report_cb)

jakkra, 2021-11-09:
**Describe the bug**
The RSSI in `bt_le_per_adv_sync_recv_info` and  `bt_df_per_adv_sync_iq_samples_report` is always -127.

**To Reproduce**
Steps to reproduce the behavior:
1. Build direction_finding_connectionless_tx sample as an advertiser.
2. Build direction_finding_connectionless_rx sample as a scanned.
3. Run both.
4. Check RSSI printed on receiver side.

**Expected behavior**
A valid RSSI value that is not constant.

**Impact**
Moderate, only affects periodic advertising.

**Logs and console output**
*** Booting Zephyr OS build v2.7.99-1230-gf94144ebe712  ***
```
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
Waiting for periodic advertising...success. Found periodic advertising.
Creating Periodic Advertising Sync...success.
Waiting for periodic sync...
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random) synced, Interval 0x00f0 (300 ms), PHY LE 1M
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
success. Periodic sync established.
Enable receiving of CTE...
success. CTE receive enabled.
Scan disable...Success.
Waiting for periodic sync lost...
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
CTE[0]: samples count 45, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -1270
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
CTE[0]: samples count 45, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -1270
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
CTE[0]: samples count 45, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -1270
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
CTE[0]: samples count 45, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -1270
PER_ADV_SYNC[0]: [DEVICE]: 07:EE:DD:D6:CD:0E (random), tx_power 127, RSSI -127, CTE AOA, data length 0, data:
CTE[0]: samples count 45, cte type AOA, slot durations: 2 [us], packet status CRC OK, RSSI -1270
...
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Windows
 - Toolchain (e.g Zephyr SDK, ...): Zephyr
 - Commit SHA or Version used: a08e6e1d9ad22920b7a08a366b3d66eaef7440f1

ppryga, 2021-11-10:
@jakkra thanks for reporting the issue. Fix is in PR: https://github.com/zephyrproject-rtos/zephyr/pull/40238

jakkra, 2021-11-10:
Thanks, tested the fix and it works.

closed 2021-11-11 by carlescufi