zephyrproject-rtos/zephyr #40223: test function set_aligned_clock  often hung

DariusBabrauskas, 2021-11-09:
**Describe the bug**
My code based on test function set_aligned_clock  often hung.

```
// /samples/drivers/counter/maxim_ds3231/src/main.c
static void set_aligned_clock(const struct device *ds3231)
{
	if (!IS_ENABLED(CONFIG_APP_SET_ALIGNED_CLOCK)) {
		return;
	}
	uint32_t syncclock_Hz = maxim_ds3231_syncclock_frequency(ds3231);
	uint32_t syncclock = maxim_ds3231_read_syncclock(ds3231);
	uint32_t now = 0;
	int rc = counter_get_value(ds3231, &now);
	uint32_t align_hour = now + 3600 - (now % 3600);

	struct maxim_ds3231_syncpoint sp = {
		.rtc = {
			.tv_sec = align_hour,
			.tv_nsec = (uint64_t)NSEC_PER_SEC * syncclock / syncclock_Hz,  //possible bug her  tv_nsec  must be < 1000 000 000
		},
		.syncclock = syncclock,
	};
.........
}
```

What have you tried to diagnose or workaround this issue?

I change .tv_nsec = (uint64_t)NSEC_PER_SEC * syncclock / syncclock_Hz,  into:
.tv_nsec = (NSEC_PER_SEC/1000) *( syncclock % syncclock_Hz), 

**To Reproduce**
not necessary 

**Expected behavior**

**Impact**
Thread hung, wait response, because sync timer  is set for long time. 

**Logs and console output**
not necessary 

**Environment (please complete the following information):**
 - OS: (Windows)
 - zephyr 2.5  and 2.7.99

**Additional context**

cfriedt, 2021-11-12:
It looks like the expression in the example should just be mod 1e9

DariusBabrauskas, 2021-11-13:
It seem to me, that it must .tv_nsec = (NSEC_PER_SEC/1000) *( syncclock % syncclock_Hz)
Because part of secons used to set internal timer which trigger, when part of secons became zero. At this moment  are sent i2c packet to set maxim_ds3231 chip. maxim_ds3231 internal counters is set to zero. So time is synhronized +- ms. Without this algorithm time is synhronized +-1s.

closed 2022-01-31 by github-actions[bot]