zephyrproject-rtos/zephyr #40236: -Unsinged int can't be used in condition compare with int

hakehuang, 2021-11-10:
**Describe the bug**
some of zephyr code has below pattern
```
static int cmd_vcs_vocs_offset_set(const struct shell *sh, size_t argc,
				   char **argv)
{
	int result;
	int index = strtol(argv[1], NULL, 0);
	int offset = strtol(argv[2], NULL, 0);

	if (index > CONFIG_BT_VCS_VOCS_INSTANCE_COUNT) {
		shell_error(sh, "Index out of range; 0-%u, was %u",
			    CONFIG_BT_VCS_VOCS_INSTANCE_COUNT, index);
		return -ENOEXEC;
	}

	if (offset > UINT8_MAX || offset < -UINT8_MAX) {
		shell_error(sh, "Offset shall be %d-%d, was %d",
			    -UINT8_MAX, UINT8_MAX, offset);
		return -ENOEXEC;
	}
```

and the offset < -UINT8_MAX is always true, as -UINT8_MAX will give you a big unsigned number

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**

**Impact**
code logic

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

hakehuang, 2021-11-10:
simliar issues found in https://github.com/zephyrproject-rtos/zephyr/pull/40186

Thalley, 2021-11-10:
@hakehuang Can you provide a log or a way to reproduce what the issue is? 

AFAIK, `UINT8_MAX` is effectively defined as 0xFF which should by default be a signed integer, so the comparison should be fine. 

The following code would emulate the above:
```
#include <stdio.h>

#define UINT8_MAX 0xff

int main()
{
	printf("%d %d\n", UINT8_MAX, -UINT8_MAX);

	return 0;
}
```
which prints
```
255 -255
```
when run. 

If you are using some other toolchain or compiler that provides a different result, please add that information to this issue.

hakehuang, 2021-11-10:
@Thalley , looks like it is define to int already.
if you change to below way, you will see error.
`
#define UINT8_MAX 0xFFu
`

Thalley, 2021-11-10:
> @Thalley , looks like it is define to int already. if you change to below way, you will see error. `#define UINT8_MAX 0xFFu`

I tried doing that with my above example as well, and got the same output. Simply compiled with `gcc -Wall`. 

Please add information about what toolchain and compiler you are using, and post the error you see.

Thalley, 2021-11-10:
> ```
> #include <stdio.h>
> 
> #define NSEC_PER_USEC 1000U
> #define USEC_PER_MSEC 1000U
> #define MSEC_PER_SEC 1000U
> #define NSEC_PER_SEC ((NSEC_PER_USEC) * (USEC_PER_MSEC) * (MSEC_PER_SEC))
> 
> #define UINT8_MAX 0xffU
> 
> void main(void)
> {
>         int a = 100;
> 
>         if (a <= -1000000000) {
>                 printf("\r\ninvalid less raw number\r\n");
>         }
> 
>         if (a <= -UINT8_MAX) {
>                 printf("\r\ninvalid less number\r\n");
>         }
> 
>         if (a <= -NSEC_PER_SEC) {
>                 printf("\r\ninvalid less\r\n");
>         }
>         if (a >= NSEC_PER_SEC) {
>                 printf("\r\ninvalid large\r\n");
>         }
> 
>         printf("\r\nNSEC_PER_SEC is %d\n\r", NSEC_PER_SEC);
>         printf("\r\nNSEC_PER_SEC is %u\n\r", NSEC_PER_SEC);
>         printf("\r\n-NSEC_PER_SEC is %d\n\r", -NSEC_PER_SEC);
>         printf("\r\n-NSEC_PER_SEC is %u\n\r", -NSEC_PER_SEC);
> }
> ```
> 
> and I get below log
> 
> gcc -o test test.c
> 
> ```
> invalid less number
> 
> invalid less
> 
> NSEC_PER_SEC is 1000000000
> 
> NSEC_PER_SEC is 1000000000
> 
> -NSEC_PER_SEC is -1000000000
> ```

Thanks - And you are correct that if comparing an int with an unsigned value, e.g. `#define UINT8_MAX 0xffU`, then it fails. However, as mentioned, `UINT8_MAX` is seemingly defined as `0xFF` and not `0xFFU`, so I'm not sure whether this is actual in Zephyr or not. Interestingly, `__UINT32_MAX__` is defined as an unsigned value, so for that it would be an issue. 

In any case, I'll update the code to use `BT_VOCS_MIN_OFFSET` and `BT_VOCS_MAX_OFFSET` (which it should do anyhow). Thanks for reporting this! 

But now I'm curious as to why the UINT16 and UINT8 values are signed, and the UINT32 and UINT64 values are unsigned... @npitre You seem to be one who added these; do you know why it is added as
```
#define __UINT8_MAX__ 0xff
#define __UINT8_TYPE__ unsigned char

#define __UINT16_MAX__ 0xffff
#define __UINT16_TYPE__ short unsigned int

#define __UINT32_MAX__ 0xffffffffU
#define __UINT32_TYPE__ unsigned int

#define __UINT64_MAX__ 0xffffffffffffffffULL
#define __UINT64_TYPE__ long long unsigned int
```
instead of 
```
#define __UINT8_MAX__ 0xffU
#define __UINT8_TYPE__ unsigned char

#define __UINT16_MAX__ 0xffffU
#define __UINT16_TYPE__ short unsigned int

#define __UINT32_MAX__ 0xffffffffU
#define __UINT32_TYPE__ unsigned int

#define __UINT64_MAX__ 0xffffffffffffffffULL
#define __UINT64_TYPE__ long long unsigned int
```

npitre, 2021-11-10:
> But now I'm curious as to why the UINT16 and UINT8 values are signed, and the UINT32 and UINT64 values are unsigned... @npitre You seem to be one who added these;

This is a mistake. The suffix is indeed missing.

Thalley, 2021-11-10:
> But now I'm curious as to why the UINT16 and UINT8 values are signed, and the UINT32 and UINT64 values are unsigned... @npitre You seem to be one who added these;
> This is a mistake. The suffix is indeed missing.

OK, I'll create a PR and add it :)

closed 2021-11-12 by cfriedt