zephyrproject-rtos/zephyr #40247: Timer HAL

sadapenore, 2021-11-10:
There is no support for Timer HAL in driver section.
Can you please add support for Timer HAL in driver section.
We could see good APIs for UART and GPIO in framework-zephyr\include\drivers.
But missing the timer HALs.
There is a HAL for LP timer but we need on normal timers.
We basically need it for STM32Fxxx. ( 429 to be precise ).

carlescufi, 2021-11-10:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/drivers/counter.h

closed 2021-11-10 by carlescufi

sadapenore, 2021-11-12:
Thank you for prompt response.

I tried using the counter example which uses RTC as clock source.
So I cannot change the resolution below 1 second.
I need timer interval in the order of few milliseconds to 100 micro seconds.
So that is not suitable. ( Please confirm )
I tried using 'timers1' but it is giving below error:
"C:/xxxxx/.platformio/packages/framework-zephyr/include/devicetree.h:303:40: error: 'DT_N_INST_0_timers1_P_label' was not declared in this scope"
The timers1 device is already define in dts file and status is marked as okay.