zephyrproject-rtos/zephyr #40254: TF-M: BL2 signing is broken due to incompatible MCUboot version

joerchan, 2021-11-10:
**Describe the bug**
Enabling TFM_BL2 (default y) results in build errors.
This is because BL2 expects an MCUboot version that includes imgtool version 1.7.2 (minimum).

**Workaround:**
Either:
 - Install 1.7.2 in system environment. (See additional context).
 - Disable TFM_BL2.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b nrf5340dk_nrf5340_cpuapp_ns ../zephyr/samples/tfm_integration/tfm_regression_test`

**Logs and console output**
```
Traceback (most recent call last):
  File "/home/joakim/repos/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py", line 128, in <module>
    wrap()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/joakim/repos/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py", line 103, in wrap
    img = imgtool.image.Image(version=imgtool.version.decode_version(version),
TypeError: __init__() got an unexpected keyword argument 'rom_fixed'
```

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain: Any
 - Commit SHA or Version used: 2.7.0

**Additinal context:**
It is unclear why installing a newer imgtool version works, and also if this is at all an acceptable solution.
BL2 wrapper.py states that the mcuboot imgtool is preferred over the system imgtool. So why it actually picks the system imgtool is strange.
```
# Add the cwd to the path so that if there is a version of imgtool in there then
# it gets used over the system imgtool. Used so that imgtool from upstream
# mcuboot is preferred over system imgtool
```

microbuilder, 2021-11-12:
@d3zd3z Any comment on the BL2 and upstream mcuboot integration here?

d3zd3z, 2021-11-18:
> BL2 wrapper.py states that the mcuboot imgtool is preferred over the system imgtool. So why it actually picks the system imgtool is strange.

I seem to have lost this argument a long time ago. My argument has always been that the tools should _only_ use the imgtool that is in the mcuboot tree, and it shouldn't be installed at all.

Aside from that, I don't know if that would actually fix this problem, since as I understand, you are using an older mcuboot, but the TF-M code is passing an argument only accepted by the newer version of imgtool. Is my understanding correct?

Can you state which version of Zephyr you are using? Is TF-M and mcuboot coming from the west.yml file?

joerchan, 2021-11-19:
> I seem to have lost this argument a long time ago. My argument has always been that the tools should _only_ use the imgtool that is in the mcuboot tree, and it shouldn't be installed at all.
>
> Aside [...]

The imgtool that is included in the mcuboot tree is the one that is being used. And that one is not compatible with the TF-M build script wrapper.

The version tested was 2.7.0, this has been partially fixed in main branch with 031ed1859b8318ce9aad5798656816510fb5bb27 (upgrading from imgtool 1.7.0 to 1.8.0.rc2), however there is still one problem.
There is a dependency now on the command always being called with the working directory set to the mcuboot/scripts path.
This is brittle and it fails currently because we don't set this for all calls to the wrapper script. I can submit a fix for that.

PR #40505 (2021-11-21): modules: trusted-firmware-m: Fix mcuboot imgtool not found on path
Fix issue with TFM signing of images not using the correct imgtool.
The wrapper command expects the mcuboot scripts folder to be the
current working directory when called in order to find its own
version of imgtool.
Since the command is using a different current working directory
this is not found and the system imgtool is used instead.
This causes the commands to be run with 2 different version of imgtool
if the system imgtool is found and does not have any issues.
The system imgtool could not be installed or have compatibilit [... cut]
commit 13230747 (2021-11-21): modules: trusted-firmware-m: Fix mcuboot imgtool not found on path

Fix issue with TFM signing of images not using the correct imgtool.
The wrapper command expects the mcuboot scripts folder to be the
current working directory when called in order to find its own
version of imgtool.
Since the command is using a different current working directory
this is not found and the system imgtool is used instead.
This causes the commands to be run with 2 different version of imgtool
if the system imgtool is found and does not have any issues.
The system imgtool could not be installed or have compatibili [... cut]
closed 2021-11-21 by nashif