zephyrproject-rtos/zephyr #40256: websocket: the size of a websocket payload is limited

gitificial, 2021-11-10:
**Description**
I try to send a websocket payload larger than 1444 bytes. The websocket log doesn't return any error but the data never arrives at the receiver. The whole content of the payload seems to reach the ./sys/socket.h sendmsg(...) function. 

A payload with a size up to 1444 bytes is transmitted flawlessly.

My receiver is a FastAPI app running on an ASGI web server. Receiving the same websocket payload larger than 1444 bytes sent with Python websockets works without any problems.

**Expected behavior**
Sending payloads larger than 1444 bytes.

**Impact**
Cannot send payload larger than 1444 bytes.

**Logs and console output**
When the payload is larger than 1444 bytes :
<dbg> net_websocket.websocket_send_msg: (): [0x200182e0] Len 1445 TEXT/1/final

When the payload is not larger than 1444 bytes :
<dbg> net_websocket.websocket_send_msg: (): [0x200182e0] Len 1444 TEXT/1/final  

**Environment**
    OS: Ubuntu 20.04LTS
    Toolchain Zephyr SDK
    Zephyr version : 2.7.0

**Hardware**
    nucleo_f746zg

rlubos, 2021-11-23:
@jukkar Thanks for confirming, I'll try to prepare a PR with a fix.

rlubos, 2021-11-23:
@gitificial I've submitted a PR with a fix proposal, would you mind giving it a try? (https://github.com/zephyrproject-rtos/zephyr/pull/40592)

closed 2021-11-23 by nashif

gitificial, 2021-11-24:
@rlubos Hi Robert, thanks a lot for the fix! I tried it with packet payloads up to 2088 bytes. It worked flawlessly.

PR #40592 (2021-11-23): net: websockets: Fix send with large payload
In case the payload and header size exceeded the network MTU size,
`websocket_send_msg()` would only send a part of the payload,
effectively leading to erronous results if called again to send the
rest. Fix the issue, by calling `sendmsg()` in a loop internally in case
it did not manage to send the entire websocket message in a single call.

Fixes #40256