zephyrproject-rtos/zephyr #40260: Multiple C language standards for the same build (Auditable Branch)

Abramo-Bagnara, 2021-11-10:
In builds of Zephyr, part of the translation units in the project are compiled in C99 mode and part in C18 mode. This is dangerous and probably not wanted: binary compatibility may not be maintained and predefined macros are certainly different in a way that may easily lead to unexpected behavior. For this reason, the MISRA documents prescribe, as a preliminary step towards MISRA compliance, the selection of one language standard.  Given that Zephyr uses several C11 features and given that C18 is, in fact, a technical corrigendum for C11, we suggest sticking to C18.

If you agree, can someone familiar with the build system change the build configuration on the Auditable Branch so that the compiler is always used in the wanted language standard?  (We suggest not leaving that to the compiler default and using an explicit compiler option instead.)

Abramo-Bagnara, 2021-11-17:
Ping. Acting upon this is needed to efficiently continue the work on compliance with the coding guidelines.

nashif, 2021-11-20:
@tejlmand  your input here please