zephyrproject-rtos/zephyr #40267: net:sockets:tls MBEDTLS Heap optimisation and API support for chained X509 certificates in DER format

Adecy, 2021-11-10:
### 1. Context
@mrfuchs and I need to optimize our MBEDTLS heap usage in an application which gathers
- 2 TLS/TCP clients which share the same X509 client certificates chain (PEM) : 1 leaf + 3 intermediate + 2 CA (PEM)
- 1 HTTPS server : public key certificate (PEM)

As all our certificates are persistent for the duration of the connections, we would like to take advantage of the MBEDTLS function `mbedtls_x509_crt_parse_der_nocopy` which (compared to its neighbor `mbedtls_x509_crt_parse_der`) doesn't create a duplicate of the provided DER certificate in heap. See [Section #Parsing X.509 certificates without copying the raw certificate data](https://tls.mbed.org/kb/how-to/reduce-mbedtls-memory-and-storage-footprint).

Currently there are two limitations to this in Zephyr :
- Zephyr doesn't take advantage of the function `mbedtls_x509_crt_parse_der_nocopy`.
- There is an API limitation where we cannot provide a chain of DER certificates, the reason is that the credential provided using the API `tls_credential_add()` is itself provided to the [`mbedtls_x509_crt_parse()`](https://github.com/ARMmbed/mbedtls/blob/ea79ff627cf0e06e0cf7fbaf861c1b9a54c65844/include/mbedtls/x509_crt.h#L535-L536)  which is able to parse concatenated PEM certificates but cannot parse more than 1 DER certificate. The two locations I identified : [CA certificate here](https://github.com/zephyrproject-rtos/zephyr/blob/b7954c9991d753177ad5a4ae984d2ae1a5d37a12/subsys/net/lib/sockets/sockets_tls.c#L643-L644) and [server certificates here](https://github.com/zephyrproject-rtos/zephyr/blob/b7954c9991d753177ad5a4ae984d2ae1a5d37a12/subsys/net/lib/sockets/sockets_tls.c#L669-L670)

### 2. Describe the solution you'd like
We would like :
- An API which allows to provide :
  - A chain of X509 "client" certificates in DER format (credential type TLS_CREDENTIAL_SERVER_CERTIFICATE)
  - A list of X509 CA certificates in DER format (credential type TLS_CREDENTIAL_CA_CERTIFICATE)
- A **specific** or **general** configuration option which enables the `mbedtls_x509_crt_parse_der_nocopy` optimisation for these types of credentials.

In this draft PR #40255, I'm proposing a **very first** implementation of what we would need as an API and behavior.

#### A. API : Providing several DER certificates as a credential

My first idea was to add new credential types `TLS_CREDENTIAL_CA_CERTIFICATE_DER_LIST` and `TLS_CREDENTIAL_SERVER_CERTIFICATE_DER_CHAIN` which handle a list of CA certificates in DER format and a list of server certificates in DER format respectively.

These credential types are almost handled the same as the original types `TLS_CREDENTIAL_CA_CERTIFICATE` and `TLS_CREDENTIAL_SERVER_CERTIFICATE` except that the DER certificates are parsed independently and assembled in the x509 using the `mbedtls_x509_crt_parse_der_nocopy` function instead of being passed to the function `mbedtls_x509_crt_parse` which do the complete work itself (but allow less possibilities).

Here is two examples to how to handle this new credential types : 

- For server certificates :
```c
static unsigned char *der_list[] = {
  ders[0], /* pointer to leaf certificate in flash */
  ders[1], /* pointer tointermediate ca certificate 1 in flash */
  ders[2], /* pointer tointermediate ca certificate 2 in flash */
  ders[3], /* pointer tointermediate ca certificate 3 in flash */
  NULL     /* null pointer to indentify the end of the list */
};
tls_credential_add(HTTPS_SECURE_TAG, TLS_CREDENTIAL_SERVER_CERTIFICATE_DER_CHAIN,
                   der_list, sizeof(*ders));
```

- For CA certificates : 
```c
static const unsigned char *ca_der_list[] = {
  ca_baltimore_root_der,
  ca_global_root_g2_der,
  NULL
};
tls_credential_add(HTTPS_SECURE_TAG, TLS_CREDENTIAL_CA_CERTIFICATE_DER_LIST,
                   ca_der_list, MAX(sizeof(ca_baltimore_root_der), sizeof(ca_global_root_g2_der)));
```

Notes :
- I made the choice to add new credential types in order to not change the behaviour of the original types.
- One problem to me here is the way to handle each DER certificates size, as a first implementation I decided to use a fixed size for the buffers. This size is provided as argument of the `tls_credential_add()` function

#### B. Optimisation : Prevent certificate duplication in HEAP by using `mbedtls_x509_crt_parse_der_nocopy`

To make it simple, I firstly introduced a **general** Kconfig option which, if enabled, use function `mbedtls_x509_crt_parse_der_nocopy` instead of `mbedtls_x509_crt_parse_der` for all certificates provided through introduced types `TLS_CREDENTIAL_CA_CERTIFICATE_DER_LIST` and `TLS_CREDENTIAL_SERVER_CERTIFICATE_DER_CHAIN`.

Note :
- Doesn't apply to original credential types
- Problem : This is a general configuration option and it cannot be enabled or disabled for specific credentials. 

### 3. Problems of this design 
In addition to API choices that I need your opinion on, the two mains problems with this design for me are :
- (Major) Not being able to enable or disable the `_der_nocopy` for some specific credentials.
- (Minor) Not being able to provide the accurate certificate size for each DER certificate.

One (good/bad) Idea I have which could solve all at once is to use an array of descriptors (see bellow), which gather the certificates buffer pointers, the certificates sizes and if they are **volatile** (should be copied in heap : `mbedtls_x509_crt_parse_der`) or **persistent** (shouldn't be copied : `mbedtls_x509_crt_parse_der`)

```c
struct cred_der_certificate {
	/* A pointer to the DER certificate buffer */
	const void *buf;

	/* Size of the DER certificate buffer */
	size_t size;

	/* Indicates whether the der certificate should be copied to the heap or not */
	enum { CRED_VOLATILE = 0, CRED_PERSISTENT = 1 } persistent;
};
```

```c
struct cred_der_certificate der_list[] = {
   {ders[0], 1024, CRED_PERSISTENT},
   {ders[1], 1060, CRED_PERSISTENT},
   {ders[2], 980, CRED_PERSISTENT},
   {ders[3], 1002, CRED_VOLATILE},
};
        
tls_credential_add(HTTPS_SECURE_TAG, TLS_CREDENTIAL_SERVER_CERTIFICATE_DER_CHAIN,
                   der_list, ARRAY_SIZE(der_list));
```

### 4. Describe alternatives you've considered
There is no actual alternative I think, both problematics come together as we need to provide certificates in DER format in order use function `mbedtls_x509_crt_parse_der_nocopy`. And as we always need to provide several certificates, we need a new API to do this.

### 5. Additional context
By implementing the solution described above for CA and client certificates we managed to save up to 5420B of heap during both `connect` and "TCP established state"  :
- MAXIMUM heap usage (connect) before: 36048 B after optimisation : 30628 B
- TCP established state : heap usage before : 24076 B after optimisation : 18632 B

### Conclusion

All this is technically very feasible but needs a good API, to me this enhancement enables optimisations for most of the secure applications of this kind which keep their certificates (in FLASH for example) for the entire connection duration. Moreover it helps to precisely size heap needs for secure applications.

I am attentive to any comment, idea or advice. Thank you.

mrfuchs, 2021-11-16:
@rlubos, what is your take on this? It looks like this feature has to be added in the [tls_credentials](/zephyrproject-rtos/zephyr/blob/main/include/net/tls_credentials.h) module you authored. It would be great if you had some time to review @Adecy's proposal.

rlubos, 2021-11-16:
Hi, thank you for describing the issue clearly.

#### Let's focus on the certificate chains first. 

The general idea behind this was to register multiple certificates with different secure tags, and then feed the TLS socket with an array of tags. Unless I miss something, this approach should already work out of the box with CA certificates, I see however there might be a problem configuring a certificate chain for own certificate since currently it's expected the certificate is coupled with private key. I think we could relax this requirement a bit, i. e. by default just parse the certificate (and thus add it to the `tls->own_cert` chain), and register own certificate in a similar way we [register a CA chain ](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets_tls.c#L816), after going through all of the tags.

I'm not really in favor of increasing the number of credential types with more specialized types - IMO it might be confusing to the users on which credential type to use in which situation (for instance whether to use `TLS_CREDENTIAL_CA_CERTIFICATE` or `TLS_CREDENTIAL_CA_CERTIFICATE_DER_LIST` with a single entry), I think we should strive to avoid such ambiguity.

#### Now as for the no-copy approach.

I agree this might be a nice addition to reduce heap usage. If I understand correctly though, this can only be used with DER certificates, so we need some extra work to determine if we deal with DER or PEM certificate. [mbed TLS handles this with a very simple check](https://github.com/zephyrproject-rtos/mbedtls/blob/zephyr/library/x509_crt.c#L1449) - perhaps we could provide a helper function in TLS socket implementation that'd allow to identify certificate type. In case DER certificate is detected, and no-copy option is enabled, `tls_add_ca_certificate()` could simply use `mbedtls_x509_crt_parse_der_nocopy()` instead of `mbedtls_x509_crt_parse()`.

Another question is how to enable the no-copy functionality. I see 3 options here:
1. Enable it globally with Kconfig option (I'd consider an option in TLS socket Kconfig space however, we're not really altering mbed TLS functionality here).
2. Enable it at a socket level with some new socket option - this would allow limiting the functionality only to sockets that explicitly enable it.
3. Introduce a generic wrapper structure for TLS credential, and add flags field there (something like you've already suggested, but I think the structure shouldn't be limited to a certain credential type). That way we could enable the no-copy for each credential individually.

Personally, I think 2 is a good tradeoff, with a minimal set of changes needed, but if you think the third option is more viable I think it'd be acceptable as well, it just requires more work to adapt the existing codebase to the API changes.

rlubos, 2021-11-16:
Hi, thank you for describing the issue clearly.

#### Let's focus on the certificate chains first. 

The general idea behind CA chains was to register multiple certificates with different secure tags, and then feed the TLS socket with an array of tags. Unless I miss something, this approach should already work out of the box with CA certificates, I see however there might be a problem configuring a certificate chain for own certificate since currently it's expected the certificate is coupled with private key. I think we could relax this requirement a bit, i. e. by default just parse the certificate (and thus add it to the `tls->own_cert` chain), and register own certificate in a similar way we [register a CA chain ](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets_tls.c#L816), after going through all of the tags.

I'm not really in favor of increasing the number of credential types with more specialized types - IMO it might be confusing to the users on which credential type to use in which situation (for instance whether to use `TLS_CREDENTIAL_CA_CERTIFICATE` or `TLS_CREDENTIAL_CA_CERTIFICATE_DER_LIST` with a single entry), I think we should strive to avoid such ambiguity.

#### Now as for the no-copy approach.

I agree this might be a nice addition to reduce heap usage. If I understand correctly though, this can only be used with DER certificates, so we need some extra work to determine if we deal with DER or PEM certificate. [mbed TLS handles this with a very simple check](https://github.com/zephyrproject-rtos/mbedtls/blob/zephyr/library/x509_crt.c#L1449) - perhaps we could provide a helper function in TLS socket implementation that'd allow to identify certificate type. In case DER certificate is detected, and no-copy option is enabled, `tls_add_ca_certificate()` could simply use `mbedtls_x509_crt_parse_der_nocopy()` instead of `mbedtls_x509_crt_parse()`.

Another question is how to enable the no-copy functionality. I see 3 options here:
1. Enable it globally with Kconfig option (I'd consider an option in TLS socket Kconfig space however, we're not really altering mbed TLS functionality here).
2. Enable it at a socket level with some new socket option - this would allow limiting the functionality only to sockets that explicitly enable it.
3. Introduce a generic wrapper structure for TLS credential, and add flags field there (something like you've already suggested, but I think the structure shouldn't be limited to a certain credential type). That way we could enable the no-copy for each credential individually.

Personally, I think 2 is a good tradeoff, with a minimal set of changes needed, but if you think the third option is more viable I think it'd be acceptable as well, it just requires more work to adapt the existing codebase to the API changes.

Adecy, 2021-11-17:
Thanks for your comments and suggestions @rlubos.

> The general idea behind CA chains was to register multiple certificates with different secure tags, and then feed the TLS socket with an array of tags. Unless I miss something, this approach should already work out of the box with CA certificates

You're right, I missed that.

We discussed @mrfuchs and I agree that your option 2 is a good solution for us. It fits our needs and as you pointed out it requires only "few" changes. Option 3 could be another future enhancement.

To illustrate, I added a commit https://github.com/zephyrproject-rtos/zephyr/pull/40255/commits/1e269f98dbfc577c38318e733e2dfc1651a079f6 to the draft PR #40255 with a implementation of how I understood the option 2 (that I tested with our application) : 
- Removed the 2 credentials types I firstly added
- Added the TLS socket option "TLS_DER_PERSISTANCE" which can have values VOLATILE=0 (default) or PERSISTANT=1. 
  - If set, the option is effective only when parsing DER certificates, if parsing a PEM certificate the option is ignored. Could this be ambiguous to the developper ? We wondered if it would make sense to reject PEM certificates if the option is set ?
  - This new option also only apply for credentials of type TLS_CREDENTIAL_CA_CERTIFICATE and TLS_CREDENTIAL_SERVER_CERTIFICATE.
  - A consequence is that this requires a new member in the "tls_context" structure 
- As we are using this feature in the context of the MQTT client, I added support for this new option in it.
- As private key and server certificate are handled at the same time in the current implementation, I changed it to parse the private key independently when it appears when iterating the tags. And server certificates are appended in a chain the same way as CA certificates.
  - Another consequence is that we need as many tags as we have certificates in the chain. (CA or own certificate)

What is your opinion on this design ?

rlubos, 2021-11-17:
Hi, thanks for the update, the current proposal corresponds pretty well with what I had in mind, just a few comments.

> Added the TLS socket option "TLS_DER_PERSISTANCE" which can have values VOLATILE=0 (default) or PERSISTANT=1.
If set, the option is effective only when parsing DER certificates, if parsing a PEM certificate the option is ignored. Could this be ambiguous to the developper ? We wondered if it would make sense to reject PEM certificates if the option is set ?

Up for discussion (this might only be my personal preference) but I'm wondering about the naming here. For me `TLS_DER_PERSISTANCE` is a bit cryptic in its meaning, I've been rather tinking on following the mbed TLS nomenclature here and providing something like `TLS_CERT_NO_COPY`. Theoretically, all certificates provided in `tls_credentials` storage should be considered non-volatile (at least until they're removed from the storage).

As for the PEM certificates, personally, I'd just ignore the option if set. We could document it explicitly in the option description, that the socket will try to avoid extra copying **if possible**, not giving guarantees this will actually happen. In the end we're constrained with what mbed TLS allows us to do here. Rejecting PEM in such case would also put an extra limitation on the impelmentation - preventing from mixing DER and PEM certificates on a socket (not likely, but I guess it still should be possible).

> As private key and server certificate are handled at the same time in the current implementation, I changed it to parse the private key independently when it appears when iterating the tags.

Just to clarify, in the comment in the code you wrote:
```
		/* private key is not set anymore with a server certificate,
		 *  there could have several DER server certificates for a
		 *  single private key
		 */
```

I don't think this is correct. The private key is still related only to a single server certificate (the one that holds the corresponding public key), other intermediate certificates in the chain are only provided to allow the other side to authenticate the "leaf" certificate. So it should still be expected that a single tag corresponds to both, cert and private key. But I guess it should still work with the current implementation, as we have this credential iterator that will fetch all of the credentials using the same tag.

> Another consequence is that we need as many tags as we have certificates in the chain. (CA or own certificate)

I don't think this is an issue? The idea being tags was that single tag corresponds to a single credential (or to a credential parir in case of private key/public cert for instance). You can then create certificate chains by providin a list of tags to use, this way you can re-use certificate in different chains w/o a need to register each chain explicitly in `tls_credentials`.

So to summarize, I think the current proposal looks pretty decent, and I think we could proceed with the actual PR (non-draft) after a small cleanup (TODOs etc.)

Adecy, 2021-11-18:
Hi, thanks for your feedback @rlubos 

I updated the PR (as "non-draft" now) #40255, feel free to request/make changes at your convenience (I'm not yet very confident with PR but I assume this is possible ?).

> So it should still be expected that a single tag corresponds to both, cert and private key. But I guess it should still work with the current implementation, as we have this credential iterator that will fetch all of the credentials using the same tag.

With the current implementation, the private key would be expected with the certificate for each tags. So the function will fail to add an intermediate certificate (which isn't provided with a private key).

https://github.com/zephyrproject-rtos/zephyr/blob/525fa76f4dd13d42549261c9fc846a150ee291bc/subsys/net/lib/sockets/sockets_tls.c#L738-L747

In my updated PR the function `tls_mbedtls_set_credentials()` only checks at the end that the private key is set if the own certificate is provided. However, I'm wondering if we could not just trust mbedTLS returned error in this case, which is : https://github.com/zephyrproject-rtos/mbedtls/blob/3e3e58a92de76069730c63e0d524f40fea948a61/include/mbedtls/pk.h#L54

>> Another consequence is that we need as many tags as we have certificates in the chain. (CA or own certificate)

> I don't think this is an issue? The idea being tags was that single tag corresponds to a single credential (or to a credential parir in case of private key/public cert for instance). You can then [...]

It was just to point it out, your argument is good, i'll take it

closed 2023-09-17 by lucasdietrich