zephyrproject-rtos/zephyr #40275: Why the start-up process of zephry is so fucking long?

sudashannon, 2021-11-11:
Forgive me for my title.
I just want to use a new embedded RTOS. Should I install a lot of python modules?
On my opinion, the build framework of zephyr is too fat to use.

sudashannon, 2021-11-11:
- An user defined toolchain and build framework
- A module-configured method
- All code

That is what I need.

carlescufi, 2021-11-11:
>I just want to use a new embedded RTOS. Should I install a lot of python modules?

There are several types of RTOS. If all you want is a simple kernel with no build system then there are other options available. Zephyr is a complex project, so it needs a complete build and configuration system, and Python is used for parts of that.

carlescufi, 2021-11-11:
I will convert this to a discussion in the meantime.

closed 2021-11-11 by carlescufi