zephyrproject-rtos/zephyr #40298: Bluetooth assertions in lll_conn.c

xudongzheng, 2021-11-11:
I have a project running on nRF52840, currently using Adafruit nRF52840 Feather. Zephyr is currently at 2.7.0 (3f826560aaf81a444018293bd6acce3c339fe150).

The development board in question acts as central to several devices and peripheral to several others. I2C is configured for DS3231 and SPI is configured for NOR flash but neither is used frequently. In fact, neither should be in use outside of at boot time. External flash is used for settings so possible that it's in use when the assertion occurs but I'm not sure.

Without any special configurations, I will get this assertion failure typically after minutes/hours.

```
ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:324
[00:24:45.255,065] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:24:45.255,096] <err> os: r3/a4:  0x200023b8 r12/ip:  0x2001a364 r14/lr:  0x0002e3bd
[00:24:45.255,096] <err> os:  xpsr:  0x410f0011
[00:24:45.255,096] <err> os: Faulting instruction address (r15/pc): 0x0002e3c8
[00:24:45.255,096] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:24:45.255,126] <err> os: Fault during interrupt handling

[00:24:45.255,126] <err> os: Current thread: 0x20008618 (unknown)
[00:24:45.532,958] <err> os: Halting system
```

I've tried adding `CONFIG_BT_CTLR_ADVANCED_FEATURES=y` and `CONFIG_BT_CTLR_ZLI=y` as suggested in #38279 and several other issues. This results in a slightly different assertion failure (which seems to happen more quickly).

```
ASSERTION FAIL [!radio_is_ready()] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:324                                                          
[00:24:45.255,065] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002                                                                                      
[00:24:45.255,096] <err> os: r3/a4:  0x200023b8 r12/ip:  0x2001a364 r14/lr:  0x0002e3bd                                                                                      
[00:24:45.255,096] <err> os:  xpsr:  0x410f0011                                                                                                                              
[00:24:45.255,096] <err> os: Faulting instruction address (r15/pc): 0x0002e3c8                                                                                               
[00:24:45.255,096] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0                                                                                                  
[00:24:45.255,126] <err> os: Fault during interrupt handling                                                                                                                 
                                                                                                                                                                             
[00:24:45.255,126] <err> os: Current thread: 0x20008618 (unknown)                                                                                                            
[00:24:45.532,958] <err> os: Halting system      
```

I would appreciate any suggestions on debugging/resolving this issue.

Thanks!

cvinayak, 2021-11-12:
@xudongzheng `ASSERTION FAIL [!radio_is_ready()] `  indicates high interrupt latencies which could be due to other application used interrupts or `irq_lock` use by application. I do not have a Adafruit nRF52840 Feather, please try to reproduce your scenario using a nRF52840 DK and provide me steps and sample applications to reproduce the assertion.

xudongzheng, 2021-11-22:
I'll try to reproduce the problem on different hardware.

I ended up disabling the DS3231 driver and the issue has gone away so I suspect DS3231/I2C is somewhat related to the issue.

cvinayak, 2021-11-22:
Please close this issue, and you can open a new issue if you still get the assertion with steps to reproduce on a nRF52840 DK.

closed 2021-11-22 by xudongzheng

cvinayak, 2022-03-03:
FYI, #43396