zephyrproject-rtos/zephyr #40306: ESP3 BLE transmit error

sam131208, 2021-11-12:
**Describe the bug**
I'm try to send mass data(jpg photos) to esp32 or read mass data(jpg photos) ,some times will get the message"ASSERT_PARAM(0 65535), in lld_pdu.c at line 601", and disconnected.

**Expected behavior**
 what means it, and how to avoid this message.

**Impact**
While continue read many photos from esp32 to phone, or write many photos to esp32. 

**Logs and console output**
[00:05:54.159,000] <dbg> bt_att.bt_att_chan_req_send: req 0x3fffc604
[00:05:54.159,000] <dbg> bt_att.chan_req_send: chan 0x3ffef4c0 req 0x3fffc604 len 252
--- 3 messages dropped ---
[00:05:54.160,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 4 -> 3
[00:05:54.160,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 3 -> 2
[00:05:54.160,000] <dbg> bt_hci_core.process_events: count 3
--- 50 messages dropped ---
[00:05:54.199,000] <dbg> bt_att.bt_att_chan_req_send: req 0x3fffc604
[00:05:54.199,000] <dbg> bt_att.chan_req_send: chan 0x3ffef4c0 req 0x3fffc604 len 252
--- 3 messages dropped ---
[00:05:54.199,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 4 -> 3
[00:05:54.200,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 3 -> 2
--- 52 messages dropped ---
[00:05:54.239,000] <dbg> bt_att.chan_req_send: chan 0x3ffef4c0 req 0x3fffc604 len 252
[00:05:54.240,000] <dbg> bt_att.chan_send: code 0x1d
[00:05:54.240,000] <dbg> bt_l2cap.bt_l2cap_send_cb: conn 0x3ffd06b8 cid 4 len 252
--- 1 messages dropped ---
[00:05:54.240,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 4 -> 3
[00:05:54.240,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 3 -> 2
--- 52 messages dropped ---
[00:05:54.279,000] <dbg> bt_att.chan_req_send: chan 0x3ffef4c0 req 0x3fffc604 len 252
[00:05:54.279,000] <dbg> bt_att.chan_send: code 0x1d
[00:05:54.279,000] <dbg> bt_l2cap.bt_l2cap_send_cb: conn 0x3ffd06b8 cid 4 len 252
--- 1 messages dropped ---
[00:05:54.279,000] <dbg> bt_conn.bt_conn_unref: handle 0 ref 4 -> 3
[00:05:54.280,000] <dbg> bASSERT_PARAM(0 65535), in lld_pdu.c at line 601
Disconnected (reason 8)
ASSERT_PARAM(0 9), in lld_pdu.c at line 601

**Environment (please complete the following information):**
OS: windows 11
Toolchain Zephyr SDK,
Branch: main

sylvioalves, 2021-11-16:
@sam131208 I forwarded to internal BLE team to understand the root cause. Are you able to share the code so we can reproduce in here? If no, no problem.

sam131208, 2021-11-19:
@sylvioalves  
1.  The kidsmusic_rom.bin is burn to 16MB flash（w25q128）,it include some photos that will be upload to app.
2.  Install the smarch.app  to android phone.
3.  build and flash kidsclock_ESP32t project to esp32_devkitc board.
4.  open the smarch app, paired with esp32, waiting upload photos to app.
5.  Open the "Gallery" page, and open the "user" switcher. add photos to esp32 board.
6.  Clear the smarch app 's  store data.
7.  reopen smarch, the app will upload all photos, waiting the error occurred
8.  repeat step 6 and step 7
    

[kidsclock_ESP32t.zip](https://github.com/zephyrproject-rtos/zephyr/files/7567230/kidsclock_ESP32t.zip)

[kidsmusic_rom.zip](https://github.com/zephyrproject-rtos/zephyr/files/7567238/kidsmusic_rom.zip)

[smarch.zip](https://github.com/zephyrproject-rtos/zephyr/files/7567332/smarch.zip)

carlescufi, 2022-01-26:
@sylvioalves ping

sylvioalves, 2022-03-18:
@sam131208, are you able to check whether this change #43947 fix this issue?  
i am not able to test it.

sam131208, 2022-03-26:
@sylvioalves 
Thanks! I have tested it and it works fine.
However, now I found another problem. While transferring data to the ESP32 and storing it on the flash, I sometimes get a "ATT Timeout" message. Can you improve flash write performance?

closed 2022-03-29 by sam131208