zephyrproject-rtos/zephyr #40316: Error undefined reference to `__aeabi_uldivmod' when build Zephyr 2.7.0 for STM32L476

fridaprog, 2021-11-12:
I've updated my VisualStudioCode + PlatformIO development environment to the **Zephyr 2.7.0** just released today. After this change my program cannot be builded correctly as before due to a link error:
undefined reference to `__aeabi_dcmpun'
this unresolved comes out using the sqrt() math function. Removing the call to sqrt() my program can be builded correctly.

The build log shows the folloing:

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_l476vg.html
PLATFORM: ST STM32 (15.0.0) > ST 32L476GDISCOVERY
HARDWARE: STM32L476VGT6 80MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-zephyr 2.20700.211026 (2.7.0) 
 - tool-cmake 3.21.3 
 - tool-dtc 1.4.7 
 - tool-gperf 3.0.4 
 - tool-ninja 1.7.1 
 - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
Reading CMake configuration...
Parsing system calls...
Generating syscall files
Generating KObject files...
Validating driver...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/disco_l476vg/offsets/zephyr/arch/arm/core/offsets/offsets.c.o
Compiling .pio/build/disco_l476vg/zephyr_final/zephyr/misc/empty_file.c.o
...
Archiving .pio/build/disco_l476vg/zephyr/kernel/libkernel.a
Indexing .pio/build/disco_l476vg/zephyr/kernel/libkernel.a
Linking .pio/build/disco_l476vg/zephyr/firmware-pre.elf
/home/alfredo/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.bfd: /home/alfredo/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libm.a(lib_a-w_sqrt.o): **in function `sqrt':**
w_sqrt.c:(.text.sqrt+0x26): **undefined reference to `__aeabi_dcmpun'**
collect2: error: ld returned 1 exit status
*** [.pio/build/disco_l476vg/zephyr/firmware-pre.elf] Error 1

**The source code (main.c):**

#include <zephyr.h>
#include <math.h>
void main(void)
{
    float a = sqrt(1.23f);
}

**prj.conf contains:**

CONFIG_NEWLIB_LIBC=y

**platformio.ini contains:**

[env:disco_l476vg]
platform = ststm32
board = disco_l476vg
framework = zephyr

erwango, 2021-11-16:
Note: When submitting issue in Zephyr repo, please use Zephyr target names, `stm32l476g_disco` in this case

I had a try to compile the proposed sample code on v2.7.0 code base and I faced no warning except `unused variable a`.
I suspect environment issue on PlatformIO.
I was using SDK 0.13.0, using zephyr toolchain variant.

fridaprog, 2021-11-17:
> I had a try to compile the proposed sample code on v2.7.0 code base and I faced no warning except `unused variable a`. I suspect environment issue on PlatformIO. I was using SDK 0.13.0, using zephyr toolchain variant.

If you have obtained a correct compilation & link probably the problem I'm facing is due to a wrong configuration in the PlatformIO build system. I'm not an expert of this environment, so I will try to put the same question to the PlatformIO guys. Thank you.

erwango, 2021-11-18:
ok, thanks for the heads up, I'm closing this issue.

closed 2021-11-18 by erwango