zephyrproject-rtos/zephyr #40329: nucleo_g0b1re: FDCAN message RAM write fails on byte-oriented write

dragonlock2, 2021-11-14:
**Describe the bug**
Sending a message using FDCAN hangs. Issue isolated to structs being written to message RAM byte by byte instead of word by word. Using `CONFIG_SPEED_OPTIMIZATIONS=y` fixes the issue as it optimizes `memcpy` to go word by word.

**To Reproduce**
1. Clone [RoboSub_FW](https://github.com/berkeleyauv/RoboSub_FW) into a **west** workspace.
2. Modify [tests/nucleo_g0b1re_canfd/prj.conf](https://github.com/berkeleyauv/RoboSub_FW/blob/main/tests/nucleo_g0b1re_canfd/prj.conf) to set `CONFIG_SPEED_OPTIMIZATIONS=n`.
3. Build and run [tests/nucleo_g0b1re_canfd](https://github.com/berkeleyauv/RoboSub_FW/tree/main/tests/nucleo_g0b1re_canfd).

**Expected behavior**
FDCAN transmission doesn't hang and program continues as expected.

**Impact**
FDCAN peripheral is unusable without compiler optimization setting.

**Logs and console output**
Running the following code (without optimization setting) results in `*y==0x44444444` which is invalid. Using a normal RAM address instead such as `0x20000000` results in expected behavior.
```
volatile uint32_t  x = 0x44332211;
volatile uint32_t* y = 0x4000b400; // message RAM address
memcpy(y, &x, 4); // byte-oriented write
```
The following code results in the expected `*y==0x44332211`.
```
volatile uint32_t  x = 0x44332211;
volatile uint32_t* y = 0x4000b400; // message RAM address
*y = x; // word-oriented write
```

**Environment (please complete the following information):**
 - OS: MacOS 12.0.1
 - arm-none-eabi-gcc 10.3.1 20210824
 - zephyr-v2.7.0

**Additional context**
Issue does not show up on the STM32G4. Neither the reference manual nor errata mention this behavior. The HAL uses word-oriented writes and thus never has this issue.

erwango, 2022-01-11:
@alexanderwachter Did you had time to look at this ?

alexanderwachter, 2022-01-11:
Yep. Should be fixed with #41326

dragonlock2, 2022-01-12:
Looks like it's working now. Thanks!

closed 2022-01-12 by dragonlock2