zephyrproject-rtos/zephyr #40336: ws2812 first led not working

CodingGhost, 2021-11-14:
**Describe the bug**
when using the ws2812 example, the first LED in the chain does not work.
for example, when setting the chain length to 10, the first led does nothing and then the next 10 leds are controlled.

**To Reproduce**
Steps to reproduce the behavior:
just upload the ws2812 example (tested with NRF52840)

**Expected behavior**
first led turns on / does something when using index 0

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Fedora
 - Zephyr SDK
 - commit 742412606090f62e984a55ee205f45620b95233a (latest)
 - nrf52840

mbolivar-nordic, 2021-11-15:
I can't reproduce this on nRF52-DK on today's main (5aeb809374042b73339877862e20638dd550c729), as built and flashed with:

```
west build -b nrf52dk_nrf52832 -s samples/drivers/led_ws2812/ -t flash
```

Note that there isn't a specific overlay or config file in that sample for an nRF52840-based board, so you may need to double-check your Kconfig and/or devicetree settings.

To verify that all pixels are working properly as a double-check of my own, I applied the following patch to the sample:

```diff
diff --git a/samples/drivers/led_ws2812/src/main.c b/samples/drivers/led_ws2812/src/main.c
index 62d7fec655..6094b6f46e 100644
--- a/samples/drivers/led_ws2812/src/main.c
+++ b/samples/drivers/led_ws2812/src/main.c
@@ -49,23 +49,7 @@ void main(void)
 
        LOG_INF("Displaying pattern on strip");
        while (1) {
-               memset(&pixels, 0x00, sizeof(pixels));
-               memcpy(&pixels[cursor], &colors[color], sizeof(struct led_rgb));
+               memset(&pixels, 0xff, sizeof(pixels));
                rc = led_strip_update_rgb(strip, pixels, STRIP_NUM_PIXELS);
-
-               if (rc) {
-                       LOG_ERR("couldn't update strip: %d", rc);
-               }
-
-               cursor++;
-               if (cursor >= STRIP_NUM_PIXELS) {
-                       cursor = 0;
-                       color++;
-                       if (color == ARRAY_SIZE(colors)) {
-                               color = 0;
-                       }
-               }
-
-               k_sleep(DELAY_TIME);
        }
 }
```

this lights up all the pixels. On my Adafruit NeoPixel ring, I see them all turn on:

[image]

If you are positive it's not a software problem, perhaps your hardware is broken?

CodingGhost, 2021-11-15:
Thanks for the effort!
I made the nrf52840 files by copying them from the nrf52832 config.

How would I compile for nrf52840 without doing that?
I am 100% sure its not a hardware issue as I have a compiled hex that turns on all leds.

mbolivar-nordic, 2021-11-15:
Please share exact steps to reproduce, including any overlays or conf files, and steps to build and flash, like I did above.

CodingGhost, 2021-11-19:
ok, I deleted everything and started again to make sure I didnt modify anything.
1. pull zephyr git and init west
2. git checkout zephyr-v2.7.0
3. west update
4. copy led_ws2812/boards/nrf52dk_nrf52832.overlay to led_ws2812/boards/nrf52840dk_nrf52840.overlay and add mosi-pin = <26>;
5. copy the .prj from the nrf52832 and remove the PAN58 line.
6. west build -p auto -b nrf52840dk_nrf52840 samples/drivers/led_ws2812
7. west flash

now everything works, except for the first led, but its 16 leds that are working.
I know 100% sure it is not a hardware issue.
somehow it seems to be possible to not have that problem, as I know somebody else, that compiled the program and sent me the hex, and that is working, although he modified more than me for another board. and compiled with a different toolchain.
I think it is neccessary to test this on a 840, as the 832 maybe wont behave the same.

I hope somebody can reproduce this!

mbolivar-nordic, 2021-11-20:
I'll try this when I can. I am off next week, but this is in my queue. Thanks for the detailed steps.

mbolivar-nordic, 2021-11-20:
In the meantime, I would recommend looking at the signal pin in a logic analyzer. It's very boring to count pulses for this protocol, but it can be useful. Since it's your first LED that's not working, I would try reducing your `chain-length` value to 1 to reduce the number of bits on the wire. That should make it possible to debug. 

See this excellent blog post for details on the timing: https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

CodingGhost, 2021-11-22:
@mbolivar-nordic I (accidentally) found something out:
If I set STRIP_NUM_PIXELS to chain-length-1 (for example 1 and 2) instead of the same value for both, the first LED does work!
BUT its red channel is always on, so green (only on that led) is yellow, blue purple and so on. also only red somehow is white.
writing 0x00 0x00 0x00, is still red. Its super strange.

CodingGhost, 2021-11-22:
another thing: I experimented with the zephyr code, and found out that writing at px_buf += 16 is the first led red channel, writing 0 there will turn off red.
pxbuf +=32 is the first led green channel and + 48 the blue channel. using this and spi_ser directly I can control the color of the first led, but somehow I cannot get any other LED to do something with that method (I am not really familiar with ws2812 protocol) but this all seems really odd and strange.

CodingGhost, 2021-11-22:
@mbolivar-nordic I finally found out the root cause of all this!
the problem is, that the nrf52832 overlay will use arduino_spi, which is SPIM3, and that seems to be broken on the nrf52840.
I would recommend adding a separate overlay for the nrf52840 that uses SPI1 (I tested with SPI1 and everything works fine!)
or maybe just move arduino_spi from SPI3 to SPI1 if that doesnt brake something else.

closed 2022-02-05 by github-actions[bot]