zephyrproject-rtos/zephyr #40348: Uart rx interrupt callback fails to work with LVGL

glin00, 2021-11-15:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

glin00, 2021-11-15:
To further test this issue, I also found that GPIO interrupt callback(ex.G14 pin) fails to work when I enable the lv_task_handler.  If I comment out lv_task_handler,  GPIO interrupt callback start to work.

erwango, 2021-11-15:
@glin00 This is likely a hardware pin conflicts. Can you provide details about your pin configurations ?

glin00, 2021-11-15:
/ {
	model = "STMicroelectronics STM32L496G-SMALLOLED board";
	compatible = "st,stm32l496g-smallOLED";

	chosen {
		zephyr,console = &usart2;
		zephyr,shell-uart = &usart2;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
	};

	gpio_output {
		compatible = "gpio-leds";

		green_led_1: led_1 {
			gpios = <&gpiod 11 GPIO_ACTIVE_HIGH>;
			label = "USER_LED_CTL1";
		};

		gpiospi_nss:  spi_nss{
	    		label = "gpio spi nss";
			gpios = <&gpiog 11 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
		};
		gpiospi_dcx:  spi_dcx{
			label = "gpio spi dcx";
			gpios = <&gpioi 4 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
		};
		gpiospi_reset:  spi_reset{
			label = "gpio spi reset";
			gpios = <&gpiod 13 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
		};
		
		gpiotp_reset:  tp_reset{
			label = "gpio tp reset";
			gpios = <&gpiod 14 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
		};

		gpiooled_avdden:  oled_avdden{
			label = "oled avdden";
			gpios = <&gpiod 12 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
		};

	};

	aliases {
		greenled1 = &green_led_1;
		spinss=&gpiospi_nss;
		spidcx=&gpiospi_dcx;
		spireset=&gpiospi_reset;
		tpreset=&gpiotp_reset;
		oledavdden=&gpiooled_avdden;
	};
};

&clk_hsi {
	status = "okay";
};

&pll {
	div-m = <1>;
	mul-n = <20>;
	div-p = <7>;
	div-q = <2>;
	div-r = <4>;
	clocks = <&clk_hsi>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(80)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
};

&usart1 {
	pinctrl-0 = <&usart3_tx_pc4 &usart3_rx_pc5>;
	current-speed = <115200>;
	status = "okay";
};

&usart2 {
	pinctrl-0 = <&usart2_tx_pa2 &usart2_rx_pd6>;
	current-speed = <115200>;
	status = "okay";
};

&timers2 {
	status = "okay";

	pwm2: pwm {
		status = "okay";
		pinctrl-0 = <&tim2_ch1_pa0>;
	};
};

&i2c3 {
	pinctrl-0 = <&i2c3_scl_pc0 &i2c3_sda_pc1>;
	status = "okay";
};

&spi1 {
	pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pb4 &spi1_mosi_pb5>;
	dmas = < &dma1 3 1 0x20440 &dma1 2 1 0x20480 >;
	dma-names = "tx", "rx";
	status = "okay";

	oled@3c {
        compatible = "everdisplay,e1604ac61";
        reg = <0x3c>;
        label = "SmallOLED";
    };

};

&spi1_mosi_pb5 {
	slew-rate = "very-high-speed";
};

&spi1_sck_pa5 {
	bias-pull-up;
};

&rtc {
	status = "okay";
};

&usbotg_fs {
	pinctrl-0 = <&usb_otg_fs_dm_pa11 &usb_otg_fs_dp_pa12>;
	status = "okay";
};

glin00, 2021-11-15:
So, why

glin00, 2021-11-15:
I add another test. I created a timer using K_TIMER_DEFINE which get called every 10msec. This timer also stops to work when  lv_task_handler is used.  If I comment out lv_task_handler, this timer starts to work.

glin00, 2021-11-15:
I try another test. This time I test I2C using I2C3(see the above dts file). This I2C also stops to work when lv_task_handler is used. If I comment out lv_task_handler, this I2C starts to work.

erwango, 2021-11-15:
Did you had a try to use debugger and see what happens when `lv_task_handler` is used ?

glin00, 2021-11-16:
Is there any way to debug and trace code inside lv_task_handler if I only have uart console as my debug tool? My IDE debugger(ex.Eclipse or Vscode) is not configured properly yet.

erwango, 2021-11-16:
Please note that `west debug` is available as command line tool.

glin00, 2021-11-17:
I can use west flash and west debug on my developement board STM32L496G_DISCO. But the problem is, if I switch to STM32L496AGI6 and ST-LINK/V2, which is this project's MCU, I get stuck with this error message(see attached). Do you have any idea how to solve this?

erwango, 2021-11-17:
> I can use west flash and west debug on my developement board STM32L496G_DISCO. 

And can't you reproduce the bug using this board ?

> But the problem is, if I switch to STM32L496AGI6 and ST-LINK/V2, which is this project's MCU, I get stuck with this error message(see attached). Do you have any idea how to solve this? [error.zip](https://github.com/zephyrproject-rtos/zephyr/files/7552378/error.zip)

You might want to try new openocd.cfg options, such as:
```
reset_config srst_only srst_nogate connect_assert_srst
set CONNECT_UNDER_RESET 1
set CORE_RESET 0
```

glin00, 2021-11-17:
After I turn on the log, I got the following prints from lv_task_handler. 
Program just stop to run inside lv_task_handler.

[1;33m<wrn> lvgl: lv_init: already inited[0m
[00:00:00.295,000] [0m<inf> lvgl: Object create ready[0m
[00:00:00.295,000] [0m<inf> lvgl: container created[0m
[00:00:00.295,000] [0m<inf> lvgl: button created[0m
[00:00:00.295,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:00:00.296,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:00:00.296,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:00:00.296,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:00:00.296,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:[00:00:00.297,000] [0m<inf> lvgl: <log_strdup alloc failed>[0m
[00:00:00.361,000] [1;31m<err> os: ***** BUS FAULT *****[0m
[00:00:00.361,000] [1;31m<err> os:   Instruction bus error[0m
[00:00:00.361,000] [1;31m<err> os: r0/a1:  0x00000000  r1/a2:  0x00000800  r2/a3:  0x00000001[0m
[00:00:00.361,000] [1;31m<err> os: r3/a4:  0x48001800 r12/ip:  0x0000004e r14/lr:  0x08000969[0m
[00:00:00.361,000] [1;31m<err> os:  xpsr:  0x40000000[0m
[00:00:00.361,000] [1;31m<err> os: Faulting instruction address (r15/pc): 0x3031206e[0m
[00:00:00.361,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:00.361,000] [1;31m<err> os: Current thread: 0x20000388 (unknown)[0m
[00:00:00.422,000] [1;31m<err> os: Halting system[0m

erwango, 2021-11-17:
@vanwinkeljan would you mind having a look ?

glin00, 2021-11-18:
@vanwinkeljan Can you give me some advise on how to solve this? Thanks.

vanwinkeljan, 2021-11-18:
@glin00 The only advice I can give you is to connect GDB and see where the exact problem occurs,  but from the logs it looks like you are running out of memory. You could try to increase the main stack size.
 
Note that `lv_task_handler` is upstream LVGL code so you could check if there any know issues

glin00, 2021-11-18:
Note that lv_task_handler is upstream LVGL code so you could check if there any know issues
=>where can I find the upstream LVGL code?

glin00, 2021-11-19:
@erwango 
From [#38690](https://github.com/zephyrproject-rtos/zephyr/issues/38690)
I found that the Zephyr's display driver could make this error.
Attached is my display driver code. Can you help to see if anything looks wrong to you?
Thanks.
[oled.zip](https://github.com/zephyrproject-rtos/zephyr/files/7568894/oled.zip)

vanwinkeljan, 2021-11-19:
> => where can I find the upstream LVGL code?

https://github.com/lvgl/lvgl

> => I increase the main stack size

What is the value you used and does it work if you increase it further?

Further you are sure that you have enough memory to allocate the buffer used for LVGL. 
I see in the configs above that you are allocating a static buffer the size of the display * bytes per pixel (204KB).

Also make sure you kernel heap is large enough

glin00, 2021-11-23:
@erwango 
Instead of using stm32l496_disco as a debugging platform, I would like to use west debug on STM32L496AGI6 and ST-LINK/V2.
The following is the current error message printed from west debug. I think it is very close to work. Can you give some suggestions? Thanks.

erwango, 2021-11-23:
> @erwango Instead of using stm32l496_disco as a debugging platform, I would like to use west debug on STM32L496AGI6 and ST-LINK/V2. The following is the current error message printed from west debug. I think it is very close to work. Can you give some suggestions? Thanks.

@glin00 Please do not mix topics in the issues, it does not help in reaching issue conclusion. Also, for such support, you can consider using discord or github discussion. Thanks

glin00, 2021-11-24:
@erwango, I raised up this discussion. Thanks.
https://github.com/zephyrproject-rtos/zephyr/discussions/40584

glin00, 2021-11-24:
@vanwinkeljan ,
What is the value you used and does it work if you increase it further?
Also make sure you kernel heap is large enough

CONFIG_MAIN_STACK_SIZE=32768 
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_HEAP_MEM_POOL_SIZE=65536
CONFIG_LVGL_MEM_POOL_HEAP_KERNEL=y

With the above memory configuration, the problem os: ***** BUS FAULT ***** still remain. The ram usage is now 95% since this MCU only have 320KB SRAM. 

The SRAM usage is as the following:
300*340*2=204000 (200K) for VDB
64K for heap memory
32K for main stack.

Do you have any suggestion to solve this problem? Thanks.

vanwinkeljan, 2021-11-24:
> Do you have any suggestion to solve this problem? Thanks.

The only way to have an idea what happens is to step through the code with gdb

glin00, 2021-11-24:
The only way to have an idea what happens is to step through the code with gdb

The attached is the result I run using west debug. The execution of firmware stop right after return 0 of the display driver write function oled_driver_api.write
[display_driver_write.zip](https://github.com/zephyrproject-rtos/zephyr/files/7593771/display_driver_write.zip)
.

glin00, 2021-11-25:
I found the root cause of this problem. Thank you for your support.

closed 2021-11-25 by glin00

erwango, 2021-11-25:
@glin00, Good, thanks for the heads up.

While it is still fresh, would you mind sharing what was the root cause then ?
It could be helpful for other people than may have the same problem as you and end up reading this discussion.

glin00, 2021-11-26:
Within display driver write function oled_driver_api.write or oled_write, I declare a char buff[30] with only 30 chars capacity. Then I use sprintf(buff, " xxxxxxxxxxxx") which try to write to buff more than 30 chars.  If I increase buff size, then sprintf will not overwrite to other memory and this  os: ***** BUS FAULT ***** problem will not occur.