zephyrproject-rtos/zephyr #40356: Request more information at bug submission

erwango, 2021-11-15:
**Is your enhancement proposal related to a problem? Please describe.**

It happens often that we are missing basic inputs in bug descriptions,
and some bug descriptions look like the if user just dropped the issue
just before running away from a building in fire.

In section `**Environment **`, we should also request: 
- hardware platform in use

In `**Additional context**`, instead of `Add any other context about the problem here.`, be more explicit:
Please provide any information that could be relevant to your issue, such as pin setting, ..
Also, you could details the attempts you made to solve the issue by your own..

**Describe alternatives you've considered**
Close bugs with obviously missing information

erwango, 2021-11-15:
@carlescufi Maybe you know the person I can assign this to ?

nashif, 2021-11-15:
@erwango  why dont you just submit a PR with required changes to .github/ISSUE_TEMPLATE/bug_report.md ?

PR #40636 (2021-11-24): .github: Bug report description: Request more information
Request more information at bug submission.

Fixes #40356
commit c02197ec (2021-11-24): .github: Bug report description: Request more information

Request more information at bug submission.

Fixes #40356
closed 2021-11-24 by cfriedt