zephyrproject-rtos/zephyr #40360: Error messages with the sample: Asynchronous Socket Echo Server Using select()

mbrunnen, 2021-11-15:
**Describe the bug**

I am getting error messages when creating a listening TCP-Socket. The error
messages are also appearing, when I am using the sample
`samples/net/sockets/echo_async_select`. The error messages is:
`<err> wifi_simplelink: Invalid command: 261`. The error is coming from
`simplelink_ioctl()`, which does not understand `ZFD_IOCTL_SET_LOCK`.
However, I am not sure if this error is the root of my problems in the
other project.

**To Reproduce**

1. Add the wifi shell for connecting to an AP in an overlay file
`wifi-shell-overlay.conf`:

```
CONFIG_SHELL=y
CONFIG_NET_L2_WIFI_SHELL=y
CONFIG_NET_SOCKETS_POSIX_NAMES=n
CONFIG_DEBUG=y
```

2. Build with:
```
  west build -b cc3235sf_launchxl -p always -- \
    -DOVERLAY_CONFIG=wifi-shell-overlay.conf
```
3. Flash the device with a signed development image (Uniflash).
4. Connect with a serial terminal to a wifi net and reset the device.
5. After rebooting the message is is appearing, see the log.

**Expected behavior**

I have expected that no error is coming.

**Impact**

No being able to open a listening socket is right now a showstopper.

**Logs and console output**

```
...
[00:00:03.537,000] <inf> wifi_simplelink: [WLAN EVENT] STA Connected to the AP: my-wifi-ssid, BSSID: XX:XX:XX:XX:XX:XX
[00:00:03.562,000] <inf> wifi_simplelink: [NETAPP EVENT] IP set to: IPv4=x.x.x.x, Gateway=x.x.x.x
uart:~$ *** Booting Zephyr OS build zephyr-v2.7.0  ***
Async select-based TCP echo server waits for connections on port 4242...
[00:00:06.535,000] <inf> wifi_simplelink: [NETAPP EVENT] IP Acquired: IPv6=xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx
[00:00:06.535,000] <dbg> wifi_simplelink.simplelink_iface_init: MAC Address XX:XX:XX:XX:XX:XX
[00:00:06.540,000] <err> wifi_simplelink: Invalid command: 261
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: e.g Zephyr SDK
 - Zephyr: `zephyr-v2.7.0-0-g3f826560aa`

**Additional context**

I am actually using a `CC3235S LaunchXL` board.

rlubos, 2021-11-22:
Hi @mbrunnen,

I don't have this particular hardware so I'm not able to reporduce, but as far as I understand the issue, the error from not being able to process `ZFD_IOCTL_SET_LOCK` should not be a critical one - does it really prevents the socket from listening properly? As long as the particular socket implementation does not intend to use the mutex which protects the access to the socket, it should be fine to ignore that particular command.

It could be however highly desirable to make use of that mutex in order to allow for duplex operation of a socket (i. e. to allow parallel RX/TX from different threads). Currently, the socket will be locked for each socket call, i. e. blocking `recv()`. In case some other thread wanted to send some data through the socket, it won't be possible until the `recv()` unblocks. Because of that, the native socket implementation uses `condvar` mechanism to unlock the mutex protecting `recv()` and allow such operation. This approach however is not always possible, because it requires some asynchronous notification mechanism to release the signal the condvar (for example see https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L388).

Another approach that we took with nRF91, is to simply release the mutex if parallel RX/TX is guaranteed to be safe by the underlying implementation. You can check [this commit](https://github.com/nrfconnect/sdk-nrf/commit/18369d49d3b832ff00aa61066fa49d2d09d4a402) to see how did we achieve that for nRF91 offloaded sockets.

mbrunnen, 2021-11-22:
Thank you a lot for the detailed information, I will have a closer look.

carlescufi, 2021-12-01:
@bwitherspoon any chance you could try to reproduce this from your side?

carlescufi, 2021-12-15:
Leaving this unassigned until someone familiar with TI can take it.

dkalowsk, 2022-01-25:
@yanivma can you take a look at this issue for TI boards?

closed 2022-02-22 by carlescufi