zephyrproject-rtos/zephyr #40367: sample: cycle_64 is failing out due to a timeout on 64-bit versions of qemu_x86 and ehl_crb

yerabolu, 2021-11-15:
**Describe the bug**
Sample cycle64 in Kernel should run successfully but it is failing on **64-bit version of qemu_x86 and ehl_crb**. 

What have you tried to diagnose or workaround this issue?
Ran the sample on 32-bit qemu_x86 and it is passing without any issue. 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p qemu_x86_64 -T samples/kernel/cycle64/  -j 1 -vvv    
                    **or** 
 twister -W -p ehl_crb --device-testing --device-serial /dev/ttyUSB0 -T samples/kernel/cycle64/  --west-flash=/home/ztest/EHL_X86_PXE.sh -j 1

2. See timeout.

**Expected behavior**
```
DEBUG   - QEMU (2771557): Booting from ROM..*** Booting Zephyr OS build v2.7.99-1382-g742412606090  ***
DEBUG   - QEMU (2771557): wrap-around should occur in 42s
DEBUG   - QEMU (2771557): [ddd:hh:mm:ss.0ms]
DEBUG   - QEMU (2771557): [000:00:00:00.000]: cycle: 000000000001fad0
DEBUG   - QEMU (2771557): [000:00:00:42.960]: cycle: 00000001000fd823
DEBUG   - QEMU (2771557): [000:00:01:25.920]: cycle: 00000002001f9a23
DEBUG   - QEMU (2771557): [000:00:02:08.880]: cycle: 00000003002f5c23
DEBUG   - QEMU (2771557): SUCCESS
DEBUG   - QEMU (2771557) complete (passed) after 2.0869383811950684 seconds
DEBUG   - No timeout, return code from QEMU (None): 0
DEBUG   - return code from QEMU (None): 0
```

**Logs and console output**
```
DEBUG   - QEMU (2772183): Booting from ROM..*** Booting Zephyr OS build v2.7.99-1382-g742412606090  ***
DEBUG   - QEMU (2772183): wrap-around should occur in 42s
DEBUG   - QEMU (2772183): [ddd:hh:mm:ss.0ms]
DEBUG   - QEMU (2772183): [000:00:00:00.010]: cycle: 000000000014555c
DEBUG   - QEMU (2772183): [000:00:00:42.980]: cycle: 00000001005cecd4
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: SDK 0.13.1
 - Commit SHA: 74241260

cfriedt, 2021-11-16:
Thanks - I'll take a look at it.

cfriedt, 2021-11-16:
```
```

Right - there is a reason that these platforms are not used in

cfriedt, 2021-11-19:
Reopening since I've had 2 reports of this now.

PR #40519 (2021-11-29): samples: kernel: cycle64: modify default timeout
Integration platforms for cycle64 are those that exhibit a "time warp" characteristic - such as `qemu_riscv64` where the application runs to completion almost instantly but the clock says that 14 minutes has passed.

Fast integration tests are good for CI, and so the timeout for integration tests is lowered to 5s.

However, some users have reported that the test fails when run under twister but succeeds under west for non-integration platforms. I've observed the test taking up to 86 seconds on `qemu_cortex_a53`, which is beyond the default 60s t [... cut]
commit 7ea93a88 (2021-11-29): tests: kernel: cycle64: mark test as slow

Previously cycle64 was under `samples/`. It's been moved to
`tests/` and has been marked with `slow: True` so that it
will not disrupt CI by adding excessive delays.

Fixes #40367
closed 2021-11-29 by nashif