zephyrproject-rtos/zephyr #40371: modem: uart interface does not disable TX interrupt in ISR

JordanYates, 2021-11-15:
**Describe the bug**

The UART ISR handler `modem_iface_uart_isr` in `modem_iface_uart.c` does not disable the TX interrupt when executed.
For the `uart_nrfx_uarte` serial driver at least, this results in the UARTE ISR permanently triggering after the end of the first transmission.
This locks up the system.

**To Reproduce**

Place an `esp-at` device on a UARTE port and attempt to run the `samples/net/wifi` sample.
```
&uart0 {
	compatible = "nordic,nrf-uarte";
	current-speed = <115200>;
	status = "okay";
	tx-pin = <34>;
	rx-pin = <33>;
	rts-pin = <40>;
	cts-pin = <7>;

	esp32e16 {
		compatible = "espressif,esp-at";
		label = "esp32e16";
		status = "okay";
	};
};
```
By default, the application will not reach the boot banner.
Adding the following lines to the bottom of `modem_iface_uart_isr ` resolves this issue:
```
	/* Disable the TX interrupt once tranmission is complete */
	if (uart_irq_tx_complete(ctx->iface.dev)) {
		uart_irq_tx_disable(ctx->iface.dev);
	}
```

**Expected behavior**
The interrupt should not be permanently triggered.

**Impact**
Applications fail to boot.

**Environment (please complete the following information):**
 - Zephyr v2.7.0

carlescufi, 2022-01-12:
@rerickson1 could you please take a look and send a PR if appropriate?

rerickson1, 2022-01-12:
> @rerickson1 could you please take a look and send a PR if appropriate?

I could submit a PR but have no way of testing this. @JordanYates could you submit a PR with your requested change?

JordanYates, 2022-02-01:
> I could submit a PR but have no way of testing this. @JordanYates could you submit a PR with your requested change?

Done

closed 2022-04-04 by JordanYates