zephyrproject-rtos/zephyr #40373: mec15xxevb_assy6853: k_busy_wait test is failing

yerabolu, 2021-11-16:
**Describe the bug**
k_busy_wait test is failing on Microchip

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. twister --hardware-map $HOME/mec1.file --device-testing -T tests/kernel/threads/thread_apis/
2. see error

**Expected behavior**
Test should pass

**Logs and console output**
```
START - test_k_busy_wait
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/threads/thread_apis/src/main.c:594: test_k_busy_wait: dt >= k_cyc_to_us_floor64(100) is false
FAIL - test_k_busy_wait in 0.14 seconds
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: SDK 0.13.1
 - Commit SHA:  742412606

PerMac, 2021-11-23:
This test also fails on all nrf platforms in our setup. I am rising the priority.

yerabolu, 2022-02-03:
Closing this issue as it is not reproducible.

closed 2022-02-03 by yerabolu