zephyrproject-rtos/zephyr #40374: up_squared: isr_dynamic test is failing

yerabolu, 2021-11-16:
**Describe the bug**
isr_dynamic test is failing on up_squared platform

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p up_squared --device-testing --device-serial /dev/ttyUSB1 -T tests/kernel/interrupt/  --west-flash="$HOME/up_squared_efi.sh" -j 1 -vv
2. See error

**Expected behavior**
test should pass.

**Logs and console output**
```
*** Booting Zephyr OS build v2.7.99-1382-g742412606090  ***
Running test suite interrupt_feature
===================================================================
START - test_isr_dynamic
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/interrupt/src/dynamic_isr.c:101: test_isr_dynamic: (x86_irq_funcs[TEST_IRQ_DYN_LINE] == dyn_isr && x86_irq_args[TEST_IRQ_DYN_LINE] == (void *)ISR_DYN_ARG is false)
dynamic isr did not install successfully
FAIL - test_isr_dynamic in 0.26 seconds
===================================================================
START - test_nested_isr
SKIP - test_nested_isr in 0.1 seconds
===================================================================
START - test_prevent_interruption
locking interrupts
unlocking interrupts
timer fired
PASS - test_prevent_interruption in 2.6 seconds
===================================================================
START - test_isr_regular
irq(27)=vector(64)
irq(28)=vector(65)
not testing irq enable/disable
PASS - test_isr_regular in 0.8 seconds
===================================================================
START - test_isr_offload_job_multiple
PASS - test_isr_offload_job_multiple in 0.1 seconds
===================================================================
START - test_isr_offload_job_identi
PASS - test_isr_offload_job_identi in 0.1 seconds
===================================================================
START - test_isr_offload_job
irq(49)
PASS - test_isr_offload_job in 0.1 seconds
===================================================================
START - test_direct_interrupt
SKIP - test_direct_interrupt in 0.1 seconds
===================================================================
Test suite interrupt_feature failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment:**
 - OS: Fedora
 - Toolchain: SDK 0.13.1
 - Commit SHA: 742412606

MaureenHelm, 2021-12-01:
Reassigned to @enjiamai as Intel Platforms (X86) maintainer

enjiamai, 2021-12-02:
Hi @MaureenHelm it is due to the target irq vector is hardcoding in this test cases, try to refine it by making some changes in the test cases.

commit e61e63e3 (2022-01-18): tests: interrupt: refine the interrupt testcases

This PR include 2 changes to refine the testcases:
1. Now we using IPI to trigger interrupt in testing instead of INT
   instruction, this means we don't need to hardcode the vector
   number. That can avoid some problem.

Fixes: #40374

2. Refined the test cases. Tigger interrupt by INT instruction and
   IPI cannot be masked by irq_disabled(). Unless it's a external
   interrupt, such as a timer. Now remove those incorrect part of
   these testcases.
closed 2022-01-18 by nashif