zephyrproject-rtos/zephyr #40375: Bluetooth: Controller: Move BT_EVT_MASK_LE_META_EVENT to a common place

cvinayak, 2021-11-16:
**Is your enhancement proposal related to a problem? Please describe.**
`(event_mask & BT_EVT_MASK_LE_META_EVENT)`  is check many places in the code. Would it be possible to move this checks to a single place in e.g. `encode_control` for all LE events?

**Describe the solution you'd like**
https://github.com/zephyrproject-rtos/zephyr/pull/31009/files#r736212741

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

wolterhv, 2022-01-21:
@cvinayak I see that the pull request, which you referenced as the solution to this issue in the issue description, is merged. Is this issue still open in spite of that?

cvinayak, 2024-02-13:
Not planned in the near future.

closed 2024-02-13 by cvinayak