zephyrproject-rtos/zephyr #40376: HiFIve1 failed to run tests/kernel/workq/work/

LixinGuoX, 2021-11-16:
**Describe the bug**
Failed to run tests/kernel/workq/work/ in CI for HiFive1, triggered an assertion failure.
Seems to work fine on other platforms.

**Expected behavior**
CI should be running smoothly.

**Impact**
Showstopper

**Logs and console output**
```
12:55:05.4631 [INFO] uart0: [+0.37ms host +0s virt 0.6ms virt from start]   ===================================================================
--
  | 12:55:05.4633 [INFO] uart0: [+0.22ms host +0s virt 0.6ms virt from start]   START - test_1cpu_running_flush
  | 12:55:05.4838 [INFO] uart0: [+14.9ms host +0.1ms virt 0.7ms virt from start]
  | 12:55:05.4838 [INFO] uart0: [+1.57ms host +0s virt 0.7ms virt from start]       Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:466: test_1cpu_running_flush: k_work_busy_get(&work) not equal to K_WORK_RUNNING
  | 12:55:05.4838 [INFO] uart0: [+0.17ms host +0s virt 0.7ms virt from start]
  | 12:55:05.4839 [INFO] uart0: [+0.7ms host +0s virt 0.7ms virt from start]    FAIL - test_1cpu_running_flush in 0.174 seconds
  | 12:55:05.4839 [INFO] uart0: [+0.67ms host +0s virt 0.7ms virt from start]   ===================================================================
  | 12:55:05.4839 [INFO] uart0: [+0.36ms host +0s virt 0.7ms virt from start]   START - test_1cpu_queued_cancel
  | 12:55:05.4839 [INFO] uart0: [+0.2ms host +0s virt 0.7ms virt from start]
  | 12:55:05.4839 [INFO] uart0: [+1.58ms host +0s virt 0.7ms virt from start]       Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:125: reset_counters: k_sem_take(&sync_sem, K_NO_WAIT) not equal to -EBUSY
  | 12:55:05.4840 [INFO] uart0: [+51µs host +0s virt 0.7ms virt from start]
  | 12:55:05.4898 [INFO] uart0: [+6.27ms host +0.1ms virt 0.8ms virt from start]    FAIL - test_1cpu_queued_cancel in 0.0 seconds
```

**Environment (please complete the following information):**
 - OS: CI
 - Toolchain Zephyr SDK v0.13.1

**Additional context**
https://buildkite.com/zephyr/zephyr/builds/46541

peter-mitsis, 2021-11-29:
I started debugging this. I can replicate using (after installing renode) ...

    west build -p -b hifive1 tests/kernel/workq/work -t run

What I have observed thus far is ...

1. When the assertion at line 466 fails, k_work_busy_get() is returning 0. This is suggesting that the work has finished, and is not "running".
2. Further debugging is showing that delay_handler() is unexpectedly reaching the end of the routine by the time we invoke k_work_busy_get().
3. This is particularly odd since our "main" thread is sleeping for 1 tick, and the routine delay_handler() is supposed to be sleeping for 100 msec (13 ticks).

Continuing to dig ....

peter-mitsis, 2021-11-30:
Additional observations:
"main" thread priority: -2, work queue processing is at -4.

The .config files between hifive1 (where erroneous behaviour detected) and qemu_riscv32 (test behaves correctly) do not show anything out of the ordinary.

More digging is required, but the behaviour is as if either the k_sleep() in the "main" thread did not wake up after 1 tick, or something occurred to prevent the scheduling/execution of the main thread once it woke after the k_sleep(1).

peter-mitsis, 2021-11-30:
Debug statements indicate that 24 ticks are being announced at once. This would be waking both the workq thread and the main thread. Since the workq thread is at priority -4, it takes precedence and executes first.  Now why are 24 ticks being announced at once?

peter-mitsis, 2021-11-30:
There seems to be something funny about the renode+hifive1 combination.

The main thread (priority -2) sleeps for 1 tick. However, when it sets the mtimecmp, it sets it for 2 ticks. This is expected from the code because there seems to be indications that the minimum delay should be 1000 cycles (which is more than 1 tick). 

The workq thread (priority -4) sleeps for 13 ticks. The mtimecmp seems to be set to a reasonable value. Unfortunately, when the timer ISR executes, around 24 ticks have been calculated to have elapsed. This wakes both threads at the same time and the workq thread executes first as it is of a higher priority.

Thus far, I have not been able to find a reason that explains why the timer ISR is executing so late in the game, resulting in so many extra announced ticks.

fkokosinski, 2021-12-02:
@peter-mitsis It seems that this issue is not related to Renode. I've tried running this test on HiFive1 board and it also fails.

peter-mitsis, 2021-12-02:
@fkokosinski. Thanks for testing on the actual hardware. That helps eliminate a possible failure scenario.

peter-mitsis, 2021-12-02:
Below is the sequence of events that I am seeing.

1. Begin test_1cpu_running_flush(), cycle count = 0x303a
2. Delay current thread (z_test_thread) for 1 tick. mtime comparator set to 0x3202
3. coophi_queue.thread executes, cycle count is still 0x303a
4. The delay_handler() executes and calls k_sleep() for a duration of 13 ticks.
5. Switch to z_main_thread for some as yet unknown reason
6. Switch to preempt_queue.thread and call z_sched_wait(), cycle count is still 0x303a.
7. There should be no active threads in the system at this time while we wait for the timer to expire.
8. Timer ISR handler executes, cycle count is 0x4862, announces about 24 ticks.
9. Both z_test_thread and coophi_queue.thread wake, but coophi_queue.thread executes as it is higher priority.

The expectation was that the timer ISR would execute when the mtime cycle count was 0x3202, not 0x4862.

The current working hypothesis to explain these observations is that something is that something is preventing the timer ISR from being serviced. Presumably, this would be due to interrupts being locked somewhere, but damned if I can find it yet.

peter-mitsis, 2021-12-06:
Very strange behaviour observed. An attempt to k_sleep for N ticks on the hifive1 board results in the following number of slept ticks ...
    24 * (floor(N / 24) + 1)

This creates a step function where sleeping for 1..23 ticks results in sleeping for 24 ticks, 24..47 ticks results in sleeping for 48 ticks, 48..71 ticks results in sleeping for 72 ticks, ...

peter-mitsis, 2021-12-14:
Below is a current summary of where things currently stand ....

1. The problem appears because too many ticks are being announced at once. This leads to two threads waking up at the same time instead of sequentially.
2. On my test setup that is using Renode for emulating the hifive1, ticks are announced in multiples of 24.
3. The timer's comparison register (mtimecmp) appears to be set correctly to issue an interrupt at the correct time.
4. The system behaves as expected up until the point it enters the idle state.
5. At the point in time it enters the idle state, the timer is set to trigger at the correct time, no other threads are running, the interrupts are unlocked and furthermore the correct set of interrupts are enabled.
6. Upon exiting the idle state (via an interrupt), only one interrupt has been detected and that is the timer interrupt.
7. At the moment that the timer interrupt's ISR executes, it calculates that 24 ticks have elapsed, when only 1 tick should have elapsed.
8. The calculation is made based off the values read from the timer's mtime register.
9. Were it not for the report of the test failing on hardware, I would be inclined to believe that it was a problem with Renode. As it stands, I am rather perplexed.
10. The good news is that an unrelated issue has been found in the code that calculates the timer's wakeup cycle time (mtimecmp). A PR has been submitted for that (https://github.com/zephyrproject-rtos/zephyr/pull/41002), but this original issue as yet remains unsolved.

LixinGuoX, 2021-12-14:
It is a weird issue indeed, maybe we need to exclude this board temporary to ensure CI runs smoothly.  #41134

peter-mitsis, 2022-02-08:
Near as I can tell, the hifive1 board has been discontinued (see https://www.sifive.com/boards/hifive1).

Does this issue exist on the hifive1_revb?

LixinGuoX, 2022-02-09:
No, I am not sure, I do not have those hardwares on my side. For now, this test case is skipped due to ram overflow.

nashif, 2022-02-10:
I cant reproduce this issue on main branch:

```
11:27:50.0505 [INFO] uart0: [+0.22s host +0s virt 0s virt from start] *** Booting Zephyr OS build v3.0.0-rc2-24-g8e70d041939e  ***
11:27:50.0535 [INFO] uart0: [+3.13ms host +0s virt 0s virt from start]   Running test suite work
11:27:50.0538 [INFO] uart0: [+0.25ms host +0s virt 0s virt from start]   ===================================================================
11:27:50.0626 [INFO] uart0: [+8.8ms host +0.1ms virt 0.1ms virt from start]   START - test_work_init
11:27:50.0678 [INFO] uart0: [+5.16ms host +0s virt 0.1ms virt from start]    PASS - test_work_init in 0.0 seconds
11:27:50.0681 [INFO] uart0: [+0.31ms host +0s virt 0.1ms virt from start]   ===================================================================
11:27:50.0683 [INFO] uart0: [+0.19ms host +0s virt 0.1ms virt from start]   START - test_delayable_init
11:27:50.0688 [INFO] uart0: [+0.57ms host +0s virt 0.1ms virt from start]    PASS - test_delayable_init in 0.0 seconds
11:27:50.0691 [INFO] uart0: [+0.23ms host +0s virt 0.1ms virt from start]   ===================================================================
11:27:50.0692 [INFO] uart0: [+0.13ms host +0s virt 0.1ms virt from start]   START - test_legacy_delayed_init
11:27:50.0696 [INFO] uart0: [+0.42ms host +0s virt 0.1ms virt from start]    PASS - test_legacy_delayed_init in 0.0 seconds
11:27:50.0699 [INFO] uart0: [+0.24ms host +0s virt 0.1ms virt from start]   ===================================================================
11:27:50.0709 [INFO] uart0: [+0.41ms host +0.1ms virt 0.2ms virt from start]   START - test_unstarted
11:27:50.0723 [INFO] uart0: [+2.07ms host +0s virt 0.2ms virt from start]    PASS - test_unstarted in 0.0 seconds
11:27:50.0726 [INFO] uart0: [+0.23ms host +0s virt 0.2ms virt from start]   ===================================================================
11:27:50.0727 [INFO] uart0: [+0.11ms host +0s virt 0.2ms virt from start]   START - test_queue_start
11:27:50.0743 [INFO] uart0: [+1.64ms host +0s virt 0.2ms virt from start]    PASS - test_queue_start in 0.0 seconds
11:27:50.0746 [INFO] uart0: [+0.24ms host +0s virt 0.2ms virt from start]   ===================================================================
11:27:50.0746 [INFO] uart0: [+0.1ms host +0s virt 0.2ms virt from start]   START - test_null_queue
11:27:50.0752 [INFO] uart0: [+0.54ms host +0s virt 0.2ms virt from start]    PASS - test_null_queue in 0.0 seconds
11:27:50.0754 [INFO] uart0: [+0.24ms host +0s virt 0.2ms virt from start]   ===================================================================
11:27:50.0765 [INFO] uart0: [+1.02ms host +0.1ms virt 0.3ms virt from start]   START - test_1cpu_simple_queue
11:27:50.0941 [INFO] uart0: [+17.59ms host +0s virt 0.3ms virt from start]    PASS - test_1cpu_simple_queue in 0.41 seconds
11:27:50.0943 [INFO] uart0: [+0.31ms host +0s virt 0.3ms virt from start]   ===================================================================
11:27:50.0945 [INFO] uart0: [+0.16ms host +0s virt 0.3ms virt from start]   START - test_smp_simple_queue
11:27:50.0951 [INFO] uart0: [+0.64ms host +0s virt 0.3ms virt from start]    SKIP - test_smp_simple_queue in 0.0 seconds
11:27:50.0954 [INFO] uart0: [+0.31ms host +0s virt 0.3ms virt from start]   ===================================================================
11:27:50.0956 [INFO] uart0: [+0.16ms host +0s virt 0.3ms virt from start]   START - test_1cpu_sync_queue
11:27:50.0980 [INFO] uart0: [+2.44ms host +0.1ms virt 0.4ms virt from start]    PASS - test_1cpu_sync_queue in 0.147 seconds
11:27:50.0983 [INFO] uart0: [+0.23ms host +0s virt 0.4ms virt from start]   ===================================================================
11:27:50.0990 [INFO] uart0: [+0.7ms host +0s virt 0.4ms virt from start]   START - test_1cpu_reentrant_queue
11:27:50.1008 [INFO] uart0: [+1.8ms host +0.1ms virt 0.5ms virt from start]    PASS - test_1cpu_reentrant_queue in 0.188 seconds
11:27:50.1010 [INFO] uart0: [+0.24ms host +0s virt 0.5ms virt from start]   ===================================================================
11:27:50.1011 [INFO] uart0: [+0.13ms host +0s virt 0.5ms virt from start]   START - test_1cpu_queued_flush
11:27:50.1053 [INFO] uart0: [+4.2ms host +0.1ms virt 0.6ms virt from start]    PASS - test_1cpu_queued_flush in 0.308 seconds
11:27:50.1056 [INFO] uart0: [+0.23ms host +0s virt 0.6ms virt from start]   ===================================================================
11:27:50.1057 [INFO] uart0: [+0.13ms host +0s virt 0.6ms virt from start]   START - test_1cpu_running_flush
11:27:50.1079 [INFO] uart0: [+2.22ms host +0.1ms virt 0.7ms virt from start]    PASS - test_1cpu_running_flush in 0.182 seconds
11:27:50.1081 [INFO] uart0: [+0.18ms host +0s virt 0.7ms virt from start]   ===================================================================
11:27:50.1082 [INFO] uart0: [+0.1ms host +0s virt 0.7ms virt from start]   START - test_1cpu_queued_cancel
11:27:50.1100 [INFO] uart0: [+1.79ms host +0.1ms virt 0.8ms virt from start]    PASS - test_1cpu_queued_cancel in 0.72 seconds
11:27:50.1102 [INFO] uart0: [+0.18ms host +0s virt 0.8ms virt from start]   ===================================================================
11:27:50.1103 [INFO] uart0: [+0.11ms host +0s virt 0.8ms virt from start]   START - test_1cpu_queued_cancel_sync
11:27:50.1114 [INFO] uart0: [+1.1ms host +0s virt 0.8ms virt from start]    PASS - test_1cpu_queued_cancel_sync in 0.0 seconds
[... cut ...]
11:27:50.1773 [INFO] uart0: [+0.15ms host +0s virt 3.1ms virt from start]   ===================================================================
11:27:50.1774 [INFO] uart0: [+0.11ms host +0s virt 3.1ms virt from start]   Test suite work succeeded
11:27:50.1777 [INFO] uart0: [+0.27ms host +0s virt 3.1ms virt from start]   ===================================================================
11:27:50.1781 [INFO] uart0: [+0.42ms host +0.1ms virt 3.2ms virt from start]   PROJECT EXECUTION SUCCESSFUL
11:27:59.9489 [INFO] SiFive-FE310: Machine paused.
```

peter-mitsis, 2022-02-10:
I can confirm the observation that this test is now passing on the hifive1 (requires tweaking stack sizes to get it to fit into memory as the test has grown since this issue was first filed.)

closed 2022-02-10 by peter-mitsis