zephyrproject-rtos/zephyr #40378: Bluetooth: Controller: Use defines for access_addr and crc_init array sizes

cvinayak, 2021-11-16:
**Is your enhancement proposal related to a problem? Please describe.**
`access_addr` always seems to be of size 4; that should be #defined.
`crc_init` always seem to be of size 3; that should be #defined.

https://github.com/zephyrproject-rtos/zephyr/pull/31009/files#r736229677
https://github.com/zephyrproject-rtos/zephyr/pull/31009/files#r736229386

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

wolterhv, 2022-01-21:
Aren't these already defined in `subsys/bluetooth/controller/ll_sw/pdu.h` as `PDU_ACCESS_ADDR_SIZE` and `PDU_CRC_SIZE`? 

Excerpt of `pdu.h`:

```c
/*
 * PDU fields sizes
 */

#define PDU_PREAMBLE_SIZE(phy) (phy&0x3)
#define PDU_ACCESS_ADDR_SIZE   4
#define PDU_HEADER_SIZE        2
#define PDU_MIC_SIZE           4
#define PDU_CRC_SIZE           3
#define PDU_OVERHEAD_SIZE(phy) (PDU_PREAMBLE_SIZE(phy) + \
                                PDU_ACCESS_ADDR_SIZE + \
                                PDU_HEADER_SIZE + \
                                PDU_CRC_SIZE)
```

If so, I'm happy to take care of this one as my first issue.

bradfordarroyo, 2022-03-15:
Great

webAbd, 2026-10-06:
I would like to work on this.