zephyrproject-rtos/zephyr #40393: modem: Add SMS messaging support

johnlange2, 2021-11-16:
Many modems support SMS messaging, but there is currently no standard way to use SMS messaging in Zephyr modem drivers.

Proposed solution:
Add a header file (modem_sms.h) with supporting defines and structs
Add two function pointers to `struct modem_context` for SMS send and receive
Add `sms` sub-command to modem_shell.c to interactively send and receive SMS messaging
Modem drivers will need to add SMS send and receive functions (this can be added separately)  
Support for SMS messaging should be modem-specific with optional implementation

For most modems, only a valid modem context should be needed.
(i.e., no socket must be opened before sending or receiving SMS messages)

In some situations, it may be preferable to send/receive SMS messages using an open socket.
To support this, two ioctl/fnctl commands can be defined in modem_sms.h

jukkar, 2021-11-16:
I just wonder if applications could use the BSD socket interface (sendmsg/recvmsg) to send and receive SMS messages. Any opinions about this?

johnlange2, 2021-11-17:
Thanks for this suggestion, @jukkar . Using socket functions sendmsg/recvmsg for SMS on an open socket is an interesting idea. There are a couple of considerations though... one is that recvmsg() is not in struct socket_op_vtable. That would be easy to add, but another consideration is that a couple of modem drivers (quectel-bg9x and ublox-sara-r4) use sendmsg for non-SMS purposes, so using it for SMS would likely break them.

jukkar, 2021-12-07:
Hi @johnlange2, we briefly discussed the SMS recv support API in the todays networking forum. I think we could have is a simple recv API that would allow both polling and blocking API. Just place a timeout option to the API. If the timeout value is `K_NO_WAIT`, then the recv just checks if there is anything available and returns with proper return code. If the user sets the timeout to `K_FOREVER`, then the API will wait as long as needed. And if the timeout is some arbitrary value, then the recv will wait that amount.

johnlange2, 2021-12-11:
Sounds good, will do. Thanks!

johnlange2, 2021-12-15:
I submitted a merge request for this here: https://github.com/zephyrproject-rtos/zephyr/pull/41175

johnlange2, 2024-02-08:
This issue is still relevant. My colleague, @jtbaumann, submitted a different PR, https://github.com/zephyrproject-rtos/zephyr/pull/59805, but it has also gotten closed. I believe he intends to either refresh his PR or submit a new one for SMS functionality.