zephyrproject-rtos/zephyr #40411: Xtensa xcc compile build fails with SOF application on latest Zephyr main

kv2019i, 2021-11-16:
**Describe the bug**
Build fails with recent Zephyr main with XCC compiler. Works ok still with gcc.

**To Reproduce**
# install Xtensa toolchain for Intel cAVS target (RG-2017.8-linux)
git clone https://github.com/thesofproject/sof.git
cd sof
scripts/xtensa-build-zephyr.sh tgl-h

Above checks out "main" Zephyr directly and builds SOF application on top of it.

**Expected behavior**
Build succeed.

**Impact**
Unable to build SOF. All SOF CI cases testing Zephyr fail.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain: See above
 - Commit SHA: 022bb25691fc44541e6fadf02ab669d150b073f1

**Additional context**
Bisect the issue to this commit:
```
commit f527a81dcf35e0f111f2c892ab69873b932c43a9
Author: Torbjörn Leksell <torbjorn.leksell@percepio.com>
Date:   Tue Nov 9 09:15:37 2021 +0100

    Tracing: Tracing/Kernel dependency issue fix
    
    Added forward declarations of types declared
    throughout kernel.h at the start of
    kernel.h. With this change it is possible to
    include tracing/tracing.h early in kernel.h
    and use the tracing functionality in
    kernel.h functions without compile errors
    which would result if tracing/tracing.h
    was included at the end of kernel.h
    (type dependencies).
    
    Signed-off-by: Torbjörn Leksell <torbjorn.leksell@percepio.com>
```

kv2019i, 2021-11-16:
FYI @tleksell-pe @nashif @andyross .

andyross, 2021-11-16:
Confirmed, and it is indeed an error specific to the Xtensa compiler.  That patch did indeed duplicate various callback typedefs (for header dependency reasons), so they exist twice.  But GNU is OK with that and Cadence is not: (compiler flags pasted in from a hand-built Zephyr SOF)

```
andy@gatera:~$ cat test.c
struct a { int i; };
typedef struct a a_t;
typedef struct a a_t;

int foo(struct a *s) { return s->i; }
andy@gatera:~$ /home/andy/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc -Os -ffreestanding -fno-common -g -fms-extensions -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -fgnu89-inline -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -c test.c
test.c:3: error: redefinition of typedef ‘a_t’
test.c:2: error: previous declaration of ‘a_t’ was here
```

I'm not expert enough to know which compiler is correct per which language standard, and that's probably not helpful anyway.  What we want is a way to make xcc act like gcc here, which must surely be possible (this version of xcc is just a gcc 4.2 front end with a custom code generator).

We could put the second/repeated definitions in a __XCC__ #ifndef I guess (we can't remove them, because those are where the documentation lives).

andyross, 2021-11-16:
Some fiddling with compiler flags shows:

* The same compiler emits the error with no arguments too: `xt-xcc -c test.c` behaves identically.
* GCC 9.3.0 will emit a similar error, but only when `-pedantic` is passed and only for -std=c99 and lower (c11 is apparently where this entered the language?)
* Older xcc's like this don't support -std=c11 (that only arrives with their clang toolchain)

andyross, 2021-11-16:
Oh, and:
* xcc seems to hardwire -pedantic?   Or at least I can't find a way to disable this error even though its progenitor compiler doesn't do it by default.

andyross, 2021-11-16:
Hacky fix up in the linked PR.  Not really what we want to be maintaining, but this will work to unbreak the build for now.

marc-hb, 2021-11-16:
According to #30105 C11 is not supposed to be allowed _yet_. However none of this is documented is it?

marc-hb, 2021-11-16:
> What we want is a way to make xcc act like gcc here, which must surely be possible (this version of xcc is just a gcc 4.2 front end with a custom code generator).

Zephyr has very elaborate toolchain abstractions in its CMake code so I was under the impression that it had much higher portability ambitions. Do these ambitions exclude C11 still?

tleksell-pe, 2021-11-17:
This problem does not really surprise me, as I pointed out in #40279 (and related discussion in #40112 that spawned #40279), redefining typedefs would become a problem with compilers that are strict C99. I'm still in favor of removing the typedef duplication (unless C11 is standard) and going for a solution where we move the typedefs earlier in kernel.h (without duplication) or break them out in separate headers if needed.

@andyross You have my blessing to revert #40279 and #40112 (depends on #40279) until we agree on a better solution for the typedefs. While #40279 and #40112 are nice changes, they aren't critical for Zephyr tracing and they aren't worth breaking the build system on some platforms.

kv2019i, 2021-11-17:
It seems revert is safer. C99 does explicitly forbid this, so you could have other compilers that hit this.

closed 2021-11-17 by nashif