zephyrproject-rtos/zephyr #40420: Lower-case characters in Kconfig symbol names cause obscure errors

SebastianBoe, 2021-11-17:
**Describe the bug**
If a user uses a lower-case character in the symbol name an obscure error can occur.

**To Reproduce**
Steps to reproduce the behavior:
1. Add snippet[0] to zephyr/samples/subsys/fs/littlefs/Kconfig
2. build the sample

**Expected behavior**
Either an obivous error message or a good build.

**Actual behavior**
build/zephyr/misc/generated/configs.c will contain:

```
#include <toolchain.h>

GEN_ABS_SYM_BEGIN (_ConfigAbsSyms)

#define CONFIG_SOME_SYMBOL_lower 1;
GEN_ABSOLUTE_SYM_KCONFIG(CONFIG_GPIO, 1);
GEN_ABSOLUTE_SYM_KCONFIG(CONFIG_BOARD, 1);

```

**Impact**
Accidentally putting lower case in symbol names causes confusion for users.

Our configs.c generation does not support lowercase in symbol names. We should either modify the generation to support it. Or we should enforce that symbols are lowercase. I support enforcing that symbols are lowercase.

That could for instance be done like this:

```
diff --git a/scripts/kconfig/kconfiglib.py b/scripts/kconfig/kconfiglib.py
index 14142c7b74..4e0ccd51bc 100644
--- a/scripts/kconfig/kconfiglib.py
+++ b/scripts/kconfig/kconfiglib.py
@@ -3793,6 +3793,9 @@ class Kconfig(object):
                 self._warn("{} defined without a type"
                            .format(sym.name_and_loc))
 
+            if any(c for c in sym.name if c.islower()):
+                self._warn("{} has lower-case characters"
+                           .format(sym.name_and_loc))
 
             if sym.ranges:
                 if sym.orig_type not in _INT_HEX:

```

but it would require some cleanup.

```
warning: SHIELD_ST7735R_ADA_160x128 (defined at boards/shields/st7735r/Kconfig.shield:4) has lower-case characters

warning: fpga_LOG_LEVEL_OFF (defined at subsys/logging/Kconfig.template.log_config:8) has lower-case characters

warning: fpga_LOG_LEVEL_ERR (defined at subsys/logging/Kconfig.template.log_config:11) has lower-case characters

warning: fpga_LOG_LEVEL_WRN (defined at subsys/logging/Kconfig.template.log_config:14) has lower-case characters

warning: fpga_LOG_LEVEL_INF (defined at subsys/logging/Kconfig.template.log_config:17) has lower-case characters

warning: fpga_LOG_LEVEL_DBG (defined at subsys/logging/Kconfig.template.log_config:20) has lower-case characters

warning: fpga_LOG_LEVEL (defined at subsys/logging/Kconfig.template.log_config:25) has lower-case characters

error: Aborting due to Kconfig warnings

```

snippet[0]:

config SOME_SYMBOL_lower
	bool ""
	default y

SebastianBoe, 2021-11-17:
@tejlmand FYI

tejlmand, 2021-11-17:
>  I support enforcing that symbols are uppercase.

I don't see any reason why Zephyr should put restrictions on Kconfig symbol naming that is not already in the language itself.

The Kconfig standard doesn't forbid lower case and Kconfig can be used outside of Zephyr.
This means another project could be using Kconfig already and later added to Zephyr as a module, re-using the same Kconfig files (theoretically).

So therefore I prefer to fix `configs.c` generation.

SebastianBoe, 2021-11-17:
Wouldn't lowercase break some kind of coding convention for defines ?

tejlmand, 2021-11-17:
> Wouldn't lowercase break some kind of coding convention for defines ?

yes, but that would only be for Zephyr itself.
https://www.kernel.org/doc/html/latest/process/coding-style.html#macros-enums-and-rtl

Users extending Zephyr by adding their own modules are free to follow whatever coding style they like.
Also Zephyr coding guidelines doesn't apply when we import modules, such as mbedTLS and TF-M.

So I don't think limiting kconfig in Zephyr would be the right way to ensure capitalized names.
That should be ensured in `scripts/ci/check_compliance.py` instead, but a valid proposal.

SebastianBoe, 2021-11-17:
I'm not worried about imported Kconfig.

There is no ecosystem of re-usable Kconfig, DT, and CMake code. No matter how much we would like there to be so :)

So enforcing a strict language would work and would result in the smallest, and therefore simplest language IMHO.

But this is a minor issue, we can just fix the config generation.

No problem.

tejlmand, 2021-11-17:
> I'm not worried about imported Kconfig.
> 
> There is no ecosystem of re-usable Kconfig, DT, and CMake code. No matter how much we would like there to be so :)

i'm not talking about eco-systems of re-usable code, but the fact that there are project external to Zephyr where I don't think we should impose our coding guide lines.

For example the memfault-firmware-sdk, which has a Zephyr port and can be used with Zephyr, but is not in the Zephyr manifest.
https://github.com/memfault/memfault-firmware-sdk

And their Kconfig file (which is currently Zephyr specific):
https://github.com/memfault/memfault-firmware-sdk/blob/master/ports/zephyr/Kconfig

In theory, a project like memfault, or any project, could decide to use Kconfig in general and not only when building with Zephyr.

PR #40421 (2021-11-18): cmake: kconfig: fix issue if Kconfig symbol name contains lower case
Fixes: #40420

If a Kconfig has lower case in its symbol name, then the file configs.c
is wrongly generated.

For example a Kconfig snippet like this:
```
 config FAIL_this
 	bool "Test fail"
	default y
```

will create an autoconf.h containing this:
```
 #define CONFIG_FAIL_this 1
```

but the configs.c will wrongly contain the same
```
 #define CONFIG_FAIL_this 1
```

instead of:
```
 GEN_ABSOLUTE_SYM_KCONFIG(CONFIG_FAIL_this, 1);
```

which results in following error at compile time
```
.../build/zephyr/misc/generated/conf [... cut]
commit 52301d97 (2021-11-18): cmake: kconfig: fix issue if Kconfig symbol name contains lower case

Fixes: #40420

If a Kconfig has lower case in its symbol name, then the file configs.c
is wrongly generated.

For example a Kconfig snippet like this:
> config FAIL_this
> 	bool "Test fail"
> 	default y

will create an autoconf.h containing this:
> #define CONFIG_FAIL_this 1

but the configs.c will wrongly contain the same
> #define CONFIG_FAIL_this 1

instead of:
> GEN_ABSOLUTE_SYM_KCONFIG(CONFIG_FAIL_this, 1);

which results in following error at compile time
.../build/zephyr/misc/generated/configs.c: In function '_ConfigA [... cut]
closed 2021-11-18 by carlescufi