zephyrproject-rtos/zephyr #40433: Unable to run program with large global variable

xudongzheng, 2021-11-17:
**Describe the bug**
I'm not sure if I'm running into a bug somewhere or if I just missed some configuration option.

`src/main.c`

```
#include <zephyr.h>
#include <sys/printk.h>

struct hid_delta {
	int64_t a;
	int64_t b;
	int64_t c;
	int64_t d;
	int64_t e;
	int64_t t;
};

// Define buffer for HID deltas.
#define HID_DELTA_COUNT 4096
static struct hid_delta hid_delta_buf[HID_DELTA_COUNT];

void my_timer_handler(struct k_timer *dummy) {
	int count = 0;
	for (int i = 0; i < HID_DELTA_COUNT; i++) {
		struct hid_delta *delta = &hid_delta_buf[i];
		delta->t++;
		count++;
	}
	printk("Hello World! %s count %d\n", CONFIG_BOARD, count);
}

K_TIMER_DEFINE(my_timer, my_timer_handler, NULL);

void main(void) {
	k_timer_start(&my_timer, K_SECONDS(1), K_SECONDS(1));
}
```

`prj.conf`

```
CONFIG_USE_SEGGER_RTT=y

CONFIG_DEBUG=y
CONFIG_LOG=y
CONFIG_SEGGER_RTT_BUFFER_SIZE_UP=8192
CONFIG_LOG_BUFFER_SIZE=8192
CONFIG_LOG_STRDUP_BUF_COUNT=32

CONFIG_ASSERT=y
CONFIG_CONSOLE=y
CONFIG_RTT_CONSOLE=y

CONFIG_LOG_PROCESS_THREAD_SLEEP_MS=100

```

**Expected behavior**
I'm running this on the Nordic nRF52840 Dongle and would expect this to run but it doesn't. If I decrease HID_DELTA_COUNT to 128, this runs without a problem.

**Environment (please complete the following information):**
 - OS: Debian
 - Toolchain: gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (15:7-2018-q2-6) 
 - Commit SHA or Version used: v2.7.0 3f826560aaf81a444018293bd6acce3c339fe150

lsangild, 2021-11-18:
Now you don't present any error logs or anything, so it's a bit hard.
I had a similar issue, and maybe increasing the stack for the main thread helps.

`CONFIG_MAIN_STACK_SIZE=XXXX`
I use 4096 for stack size, but you may need some more.

xudongzheng, 2021-11-19:
> Now you don't present any error logs or anything, so it's a bit hard. I had a similar issue, and maybe increasing the stack (default 1024 bytes) for the main thread helps.
> 
> `CONFIG_MAIN_STACK_SIZE=XXXX` I use 4096 for stack size, but you may need some more.

Thank you for the suggestion. Unfortunately I get the same issue even if I increase `CONFIG_MAIN_STACK_SIZE` from 4096 to 40960.

When I increase the variable size, I don't get any error message as it doesn't even boot (I don't even get the standard `Booting Zephyr OS build zephyr-v2.7.0` line).

xudongzheng, 2021-11-27:
Upon further investigation, the program runs fine but RTT output fails to work. It gets stuck on "Searching for RTT Control Block".

carlescufi, 2021-11-30:
@xudongzheng could you please try with a more modern version of GNU Arm Embedded? Also what board are you using?

pkral78, 2021-11-30:
Without reproducible settings and build output I may only guess what is going on, but try thoes:

1.) run gdb and break on main and examine RTT control block if it contain string `SEGGER RTT` if not then control block may be stripped out or overwritten
2.) use JLink RTT directive to set fixed address and point JLink to that address, you have to update linker script to ensure that RTT block is linked to that address, autosearching control block sometimes does not work

xudongzheng, 2021-12-04:
I've created a repository to reproduce this bug: https://github.com/xudongzheng/zephyr-rtt-bug. The binary can be built with command `west build -b nrf52840dongle_nrf52840`.

I've reproduced this issue with a different gcc version on macOS `arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)`.

carlescufi, 2021-12-20:
@xudongzheng thanks. I assume you use an external J-Link debugger with that board, right?

carlescufi, 2021-12-20:
I tried [your sample](https://github.com/xudongzheng/zephyr-rtt-bug) on an nRF52840 DK and could not reproduce the issue:
```
*** Booting Zephyr OS build v2.7.99-2573-g9778f0cd088f  ***
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
Hello World! nrf52840dk_nrf52840 count 4096
```

Maybe there's something else that we are missing, but the timer callback is invoked on ISR context, and so you are calling `printk()` from ISR, which is probably not a great idea.

carlescufi, 2021-12-20:
Please reopen if you think this is indeed a bug.

closed 2021-12-20 by carlescufi

xudongzheng, 2021-12-21:
Thank you for trying to reproduce this issue. After getting the same problem on my nRF5340 DK, I realized that the issue might be how I was viewing the logs. It turns out that I needed to specify the full device name `NRF52840_XXAA` rather than just `NRF52`.

```
JLinkRTTLogger -Device NRF52840_XXAA -If SWD -Speed 4000 -RTTChannel 0 ~/Desktop/rtt.log
```