zephyrproject-rtos/zephyr #40438: Ill-formed sources due to external linkage inline functions calling static functions

Abramo-Bagnara, 2021-11-17:
In the Zephyr sources there are a few instances of static functions
used in inline functions with external linkage, which is ill-formed
for the C99 and later standards.  An example is static function
`sys_dlist_peek_head(sys_dlist_t*)', which is used by
void z_priq_dumb_add(sys_dlist_t *pq, struct k_thread *thread),
an always-inline function with external linkage.
To make the source well defined, the solution is to amend the definition
in kernel/sched.c from

ALWAYS_INLINE void z_priq_dumb_add(sys_dlist_t *pq, struct k_thread *thread)

to

static ALWAYS_INLINE void z_priq_dumb_add(sys_dlist_t *pq, struct k_thread *thread)

However, this would conflict with the (pointless?) declaration of the
function in include/kernel/sched_priq.h.  I think it is pointless because
z_priq_dumb_add() is only used in kernel/sched.c.  Thus a complete solution,
for this and a few other cases, would be to change the definitions
including the static storage class to the affected inline definitions
and remove the pointless declarations.

Please note that making the project well-formed as specified in the C
standards is a prerequisite for any form of MISRA compliance.

peter-mitsis, 2021-11-29:
I did a search for ALWAYS_INLINE, and have found that in addition to z_priq_dumb_add() the following routines had declarations in sched_priq.h, but were only used in one file.
   - z_priq_mq_add()
   - z_priq_mq_remove()

As such, those two routines are also candidates for the same correction.

However, there is an additional routine about which I am not certain if it falls under the scope of this issue. The routine in question is z_unpend_thread_no_timeout(). This routine, defined in sched.c and using ALWAYS_INLINE, is declared in ksched.h but does not use ALWAYS_INLINE. (It's use is not confined to sched.c.) The question to which I am unsure of the answer is "Should the declaration for this routine also use ALWAYS_INLINE?".

Please advise.

peter-mitsis, 2021-11-30:
Unless otherwise indicated, I'll leave z_unpned_thread_no_timeout() as is.

peter-mitsis, 2022-02-08:
The work for this has been done and already merged into main. See PR #41071, and SHA ID f8b76f3b03dec540a6aeee7b176e0f01ef2b572f

Closing.

PR #41003 (2021-12-10): kernel: add 'static' keyword to select routines (Auditable Branch)
Applies the 'static' keyword to the following inlined routines:
    z_priq_dumb_add()
    z_priq_mq_add()
    z_priq_mq_remove()
As those routines are only used in one place, they no longer have
externally visible declarations.

Fixes #40438
closed 2022-02-08 by peter-mitsis