zephyrproject-rtos/zephyr #40439: Undefined escape sequence: ill-formed for the C standard

Abramo-Bagnara, 2021-11-17:
#define Z_PROGBITS_SYM  "\%"

The ‘"\%"’ string literal used in “#define Z_PROGBITS_SYM "\%"” has
the syntax of a C escape sequence, but no defined meaning by the C standard.
It should be replaced with ‘"%"’ or with ‘"\\%"’ depending on whether
the intended behavior is to produce the ‘%’ or the ‘\%’ strings, respectively.

dcpleung, 2021-11-20:
?????

It is `"\%"` with the back slash already.

Abramo-Bagnara, 2021-11-20:
Sorry, I've messed up with the markup. Please see the edited note.

stephanosio, 2021-11-30:
https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/include/app_memory/app_memdomain.h#L84

cfriedt, 2021-11-30:
@nashif @dcpleung - can you please categorize this issue? Is it a bug?

It sounds as though this is for linker script generation.

d3zd3z, 2021-11-30:
I believe the fix is to just remove the backslash. It would generate a compiler warning, if the appropriate warning were enabled. However, I believe the behavior would be correct.

dcpleung, 2021-12-02:
Not sure if this is actually a bug... the Zephyr toolchains never complain about this, and it is working for generating the necessary assembly code.

RobertoBagnara, 2021-12-02:
Well, see, e.g., C18 6.4.4.4 (Character constants) and, in particular, footnote 179 where it says "If any other character follows a backslash, the result is not a token and a diagnostic is required."
The support for \% as a valid escape sequence could be offered as an extension by all Zephyr toolchain but, unless I am missing something, they are all GCC-based, and GCC does not officially offer such an extension: https://gcc.gnu.org/onlinedocs/gcc/C-Extensions.html#C-Extensions
As a result, Zephyr is now counting on what, at best, is an undocumented GCC extension, and GCC is not compliant to the C standard if it is true that it is not giving the diagnostic required by the standard.
Given that there is no need whatsoever to have that backslash, removing it seems to me the only sensible thing to be done.

dcpleung, 2021-12-02:
I actually have submitted a PR to remove the slash.

Just out of curiosity, what does C99 say about this? I was only able to search for the C17 draft which doesn't have footnote 179. And the C99 draft doesn't have similar footnote.

RobertoBagnara, 2021-12-02:
C99 TC3 says the same thing, as far as I can see. The clause has the same number  6.4.4.4 (Character constants), the footnote text seems the same, but the footnote number if 65.  You can download the last draft before ISO blessing here: http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf

PR #40865 (2021-12-05): kernel: mem_domain: remove extra slash in Z_PROGBITS_SYM
The macro Z_PROGBITS_SYM has "\%" but this is not correct
usage to escape "%". So remove the extra slash. Note that
this macro is used directly to generate assembly code
so it cannot be "%%" as in escaping "%" in printf().

Fixes #40439
commit d387c863 (2021-12-05): kernel: mem_domain: remove extra slash in Z_PROGBITS_SYM

The macro Z_PROGBITS_SYM has "\%" but this is not correct
usage to escape "%". So remove the extra slash. Note that
this macro is used directly to generate assembly code
so it cannot be "%%" as in escaping "%" in printf().

Fixes #40439
closed 2021-12-05 by nashif