zephyrproject-rtos/zephyr #40444: Late C++ constructor initialization on native posix boards

wcappelle, 2021-11-18:
Within zephyr the C++ constructors are delayed up to the point the hardware is initialized & working (threads,locks,devices). But when building for native posix, the behavior changes, and constructors are executed before the drivers & internal threads are initialized (the normal behavior actually). This however leads to different application behavior, and more often to segfaults in zephyr code, due to calls which are not ready yet.

A Workarround/aternaltive way of avoiding the problem can be done by postponing initialization outside of the constructor, but it's not consistent with non native posix boards.

My question now if of this is intended, or if the late constructor init can be made consistent with the non posix boards behavior?

wojciechslenska, 2022-06-13:
This merge request https://github.com/zephyrproject-rtos/zephyr/pull/46357 fixes this behavior, and now it is working the same as on hardware. More information here https://github.com/zephyrproject-rtos/zephyr/issues/39347

wcappelle, 2022-06-13:
@wojciechslenska , thanks for the effort! I was indeed following both conversations.