zephyrproject-rtos/zephyr #40447: wifi: esp_at: doesn't actually support AT v1.7's DNS query

ycsin, 2021-11-18:
The esp8266 AT driver doesn't actually supports DNS query (4431ae5119ca08b69839e3623ea8de0e483ccf86) on [AT commands v1.7](https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf).

Current implementation is based on AT v2.0, for v1.7:
1. dns command is `AT+CIPDNS_CUR` instead of `AT+CIPDNS`.
2. the returned command also seems to be different, not `+CIPDNS:enable[,"DNS IP1"[,"DNS IP2"[,"DNS IP3"]]]`

[image]

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

ycsin, 2021-11-18:
@mniestroj any idea?

mniestroj, 2021-11-22:
@ycsin I was not testing that with ESP-AT 1.7 and have little interest in implementing DNS query with 1.7 myself (besides keeping old functionality still working). But any pull requests are welcome!

ycsin, 2021-11-22:
@mniestroj thanks for the reply. Its response seems to be in multiple lines, I'm not sure how to handle the second one and so on, but I'll submit a PR if I have something, at least disabling that DNS option when 1.7 is selected so that there's no surprise.

jukkar, 2024-02-08:
There does not seem to be anyone in maintainers file that would handle esp_at wifi driver, but I found in codeowners file the driver so assigning to @mniestroj