zephyrproject-rtos/zephyr #40449: Twister tests fail when running on actual hardware due to deprecated command warning

lairdjm, 2021-11-18:
**Describe the bug**
Whilst trying twisted on a live development board with the kernel tests, they seem to fail:
```
./scripts/twister --device-testing --hardware-map map.yml -T tests/kernel
ZEPHYR_BASE unset, using "/tmp/AA/zephyr"
INFO    - Zephyr version: v2.7.99-1488-g525fa76f4dd1
INFO    - JOBS: 16
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 163 test scenarios (163 configurations) selected, 68 configurations discarded due to filters.

Device testing on:

| Platform   |           ID | Serial device   |
|------------|--------------|-----------------|
| bl653_dvk  | 000483068213 | /dev/ttyUSB0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

ERROR   - bl653_dvk                 tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout/device.log
INFO    - Total complete:    1/  95   1%  skipped:   68, failed:    1
ERROR   - bl653_dvk                 tests/kernel/condvar/condvar_api/kernel.condvar    FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/condvar/condvar_api/kernel.condvar/device.log
INFO    - Total complete:    2/  95   2%  skipped:   68, failed:    2
ERROR   - bl653_dvk                 tests/kernel/sched/deadline/kernel.scheduler.deadline FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/sched/deadline/kernel.scheduler.deadline/device.log
INFO    - Total complete:    3/  95   3%  skipped:   68, failed:    3
ERROR   - bl653_dvk                 tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap/device.log
INFO    - Total complete:    4/  95   4%  skipped:   68, failed:    4
```
The contents of the error logs is:
```
WARNING: runners: Argument --snr is deprecated, use -i/--dev-id instead.
```
So this looks to be a false error and the snr argument needs changing to dev-id or i.

**To Reproduce**
Steps to reproduce the behavior:
As above, run twisted on actual nordic hardware for kernel tests

**Expected behavior**
Tests to run and pass

**Impact**
Tests on actual hardware do not working, pretty big issue.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 525fa76f4dd13d42549261c9fc846a150ee291bc

**Additional context**
Note that this was tested in nRF connect SDK 1.6 also and these errors were not apparent on that

lairdjm, 2021-11-18:
After replacing :
```
                        command_extra_args.append("--snr")
```
with:
```
                        command_extra_args.append("--dev-id")
```
in scripts/pylib/twister/twisterlib.py, the error is no longer emitted, though it still hasn't fixed the issue. The error file now shows a full build log but it's like it hasn't rebooted the module to run the test and as a result it hasn't received any of the output. It got to test 3 that failed and I manually pressed the board's reset button now tests have started passing, they do stop passing again with the same timeouts not long after. I've also tried adding `hw-flow-control` to the UART and that also seems to vastly improve things, so is twister resetting the module and opening the UART too late and it's lost the data that it should have captured possibly?

lairdjm, 2021-11-18:
Must have missed this line in the build log since it's not really to do with building:
```
/opt/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries:  libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```
So that might explain that. Have gone through the process of building a sdk-ng toolchain for arm with modifications to use python3.9 and it gives me a segmentation fault when using it

lairdjm, 2021-11-18:
Have built python 3.8 and moved the files into place, no more error from gdb in the logs so I assume it runs, however I am still getting failures:
```
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/  95   1%  skipped:   68, failed:    0
ERROR   - bl653_dvk                 tests/kernel/fifo/fifo_api/kernel.fifo             FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/fifo/fifo_api/kernel.fifo/build.log
INFO    - Total complete:    2/  95   2%  skipped:   68, failed:    1
ERROR   - bl653_dvk                 tests/kernel/sched/deadline/kernel.scheduler.deadline FAILED: Timeout
ERROR   - see: /tmp/AA/zephyr/twister-out/bl653_dvk/tests/kernel/sched/deadline/kernel.scheduler.deadline/build.log
INFO    - Total complete:    3/  95   3%  skipped:   68, failed:    2
```
The log doesn't show any errors so I'm a bit puzzled as to what failed, I'm assuming timeout is it didn't get a response from the module but unsure. Log file contents: https://pastebin.com/HksNrktV

lairdjm, 2021-11-18:
Seems to be a problem when using a map file, contents of my map file:
```
- connected: true
  id: 000683534322
  platform: nrf52840dk_nrf52840
  product: J-Link
  runner: jlink
  serial: /dev/ttyACM1
```
When I run using a map file with debug (`./scripts/twister --device-testing --hardware-map map.yml -T tests/kernel -vv`):
```
DEBUG   - running footprint_reports
INFO    - 0 of 96 test configurations passed (0.00%), 0 failed, 67 skipped with 0 warnings in 0.93 seconds
INFO    - In total 1128 test cases were executed, 503 skipped on 1 out of total 406 platforms (0.25%)
INFO    - 0 test configurations executed on platforms, 96 test configurations were only built.

Hardware distribution summary:

| Board               |           ID |   Counter |
|---------------------|--------------|-----------|
| nrf52840dk_nrf52840 | 000683534322 |         0 |
INFO    - Saving reports...
...
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/pending for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/pending/kernel.objects
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/pending for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/mem_slab/mslab_api for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/mem_slab/mslab_api/kernel.memory_slabs.api
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/mem_slab/mslab_api for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/fifo/fifo_timeout for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/fifo/fifo_timeout for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/mem_heap/k_heap_api for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/mem_slab/mslab_concept for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/mem_heap/k_heap_api for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/mem_slab/mslab_concept for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/fifo/fifo_api for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/fifo/fifo_api/kernel.fifo
DEBUG   - Building /tmp/AA/zephyr/tests/kernel/fifo/fifo_api for nrf52840dk_nrf52840
DEBUG   - Finished building /tmp/AA/zephyr/tests/kernel/mem_slab/mslab_threadsafe for nrf52840dk_nrf52840
DEBUG   - build test: nrf52840dk_nrf52840/tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe
...
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
(this is printed a lot, then eventually...)
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
DEBUG   - -- west flash: using runner jlink
-- runners.jlink: JLink version: 7.52d
-- runners.jlink: Flashing file: /tmp/AA/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept/zephyr/zephyr.hex

DEBUG   - halted
DEBUG   - run status: nrf52840dk_nrf52840/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept flash_error
[... 2 lines cut ...]
DEBUG   - Calling cmake with arguments: ['-B/tmp/AA/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/sched/deadline/kernel.scheduler.deadline', '-S/tmp/AA/zephyr/tests/kernel/sched/deadline', '-DEXTRA_CFLAGS="-Werror"', '-DEXTRA_AFLAGS="-Wa,--fatal-warnings', '-DEXTRA_LDFLAGS="-Wl,--fatal-warnings"', '-DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror', '-GUnix Makefiles', '-DBOARD=nrf52840dk_nrf52840']
DEBUG   - Using serial device /dev/ttyACM1 @ 115200 baud
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/tmp/AA/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/pending/kernel.objects', '--runner', 'jlink', '--tool-opt=-SelectEmuBySN  000683534322']
DEBUG   - Waiting for device nrf52840dk_nrf52840 to become available
```
When I run using a command without debug (`./scripts/twister --device-testing --device-serial /dev/ttyACM1 -p nrf52840dk_nrf52840 -T tests/kernel`):
```
Renaming output directory to /tmp/AA/zephyr/twister-out.30
[... cut ...]
INFO    - 163 test scenarios (163 configurations) selected, 67 configurations discarded due to filters.

Device testing on:

| Platform            | ID   | Serial device   |
|---------------------|------|-----------------|
| nrf52840dk_nrf52840 |      | /dev/ttyACM1    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    4/  96   4%  skipped:   67, failed:    0
```
So it works fine like that. The bottom  is using a serial device and the top with the map file is using 'counter' - where does this 'counter' come from?

PerMac, 2021-12-15:
Have you tried changing `runner` in hw map to `nrfjprog` (and having nrfjprog installed)? this is the recommended tool for nrf platforms. Your output shows `FAILED Flash error (device 21.554s)` so clearly the flashing doesn't work in your setup. The warning you were seeing (--snr depriciation) is just a warning, It is not the failure root cause.

PerMac, 2021-12-15:
When I used map like yours (with `runner: jlik`) for nrf52840dk i was getting "flash error" and timeouts. After changing to nrfjprog it passes with the command your command. The difference is that if you don't use hardware-map, nrfjprog is used as default and works out of the box. With hw map you have to manually change the runner to nrfjprog

lairdjm, 2021-12-15:
@PerMac it seems the issue here is a nordic issue, not sure if it's a silicon issue or nrfjprog issue, but when the module uses DWT for null pointer exception as per https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/nordic_nrf/Kconfig.defconfig#L45 (it used to use MWU which is why I was not encountering this issue with an earlier zephyr version), you cannot reset the module using `nrfjprog -f NRF52 --reset` or `nrfjprog -f NRF52 --debugreset`, the only reset that works is `nrfjprog -f NRF52 --pinreset`, however this does not work on our boards because the reset line of the atmel is not brought out, so at this point the module cannot be rebooted. I raised this issue on the devzone which I added you to: https://devzone.nordicsemi.com/support/279994

What would also be useful is if the log file gives the real reason for the test failing, e.g. timeout, rather than just putting the nrfjprog error.

PR #41236 (2021-12-15): Various twister bug fixes
- twister: remove lock from generated hardware map
- twister: fix baud setting for detected devices
- twister: fix documentation of baud setting
- twister: do not use deprecated arguments to nrfprog

Fixes #40449 
Fixes #40450
Fixes #39179
commit 526ee4c7 (2021-12-15): twister: do not use deprecated arguments to nrfprog

--snr is deprecated, so use --dev-id instead.

Fixes #40449
closed 2021-12-15 by nashif