zephyrproject-rtos/zephyr #40497: Problematic debug  if mcuboot enabled with signed image

DariusBabrauskas, 2021-11-19:
**Is your enhancement proposal related to a problem? Please describe.**
Impossible debug  if mcuboot enabled with signed image, because signed data is added into zephyr.signed.bin or zephyr.signed.hex file, but not in to  zephyr.elf.
So debug is problematic.

**Describe the solution you'd like**
Add sign data into   zephyr.elf file too.

**Describe alternatives you've considered**
Currently I  manually  flash zephyr.signed.hex.  After this, I try  attach with debugger, with not signed zephyr.elf.

**Additional context**

gmarull, 2021-11-19:
In case you use VSCode, I had some suggess debugging bootloader/Zephyr with this configuration:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/zephyr/zephyr.signed.hex",
            "name": "Debug Firmware",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "nrf52840_xxaa",
            "targetId": "nrf52",
            "interface": "swd",
            "runToMain": true,
            "preLaunchCommands": [
                "add-symbol-file ${workspaceRoot}/build/zephyr/zephyr.elf",
            ]
        },
        {
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build_bootloader/zephyr/zephyr.elf",
            "name": "Debug Bootloader",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "device": "nrf52840_xxaa",
            "targetId": "nrf52",
            "interface": "swd",
            "runToMain": true,
        },
    ]
}
```

DariusBabrauskas, 2021-11-20:
Thank you. 
I am using  eclipse enviroment, stlink, swd, pyocd, stm32l4a6 and additional plugins. It seems to me, that best solution for all  - to do add sing information into zephyr.elf file.

gmarull, 2024-02-13:
As mentioned, it's possible to debug using `add-symbol-file`.

closed 2024-02-13 by gmarull