zephyrproject-rtos/zephyr #40534: Type correctness of predicate macros

Abramo-Bagnara, 2021-11-21:
There are several macros in Zephyr that are clearly meant to be
predicates and evaluate to either 0 or 1. Some of them are used in
preprocessor conditionals, and thus there is no choice.  For the
others, given that MISRA C forbids mixing integers and Booleans, we
have two options:

1) Change those macros so that they evaluate to true or false.

2) Leave them as they are and then adapt their use in contexts expecting
   a Boolean as follows: MACRO() != 0.

We recommend option (1), but this is your choice.  Please note that a
decision about this should be reached as quickly as possible, as it has
a deep influence on the subsequent MISRA compliance efforts.

jfischer-no, 2021-11-22:
Is it about rule 14.4?

>Leave them as they are and then adapt their use in contexts expecting
a Boolean as follows: MACRO() != 0.

By which C standard does `MACRO() != 0` have type boolean?

Abramo-Bagnara, 2021-11-22:
It is about essential type model of MISRA (that cover other rules together with Rule 14.4).
According to it any comparison (e.g. `MACRO() != 0`) has essential type Boolean (similar in that to C++ standard type).
Of course this has nothing to do with C standard where comparison has int type.
One of the aim of MISRA essential type model is to avoid mixing data that is Boolean with data that is numeric and I think there is little doubt that the domain of a comparison is Boolean.

ElectricalPaul, 2021-11-23:
I agree with @peter-mitsis. Changing the macro to return strict bool values should require a lot less code churn; we change the macro definition, and leave the code alone everywhere the macro is used with an implicit cast from int to bool.

ceolin, 2021-11-30:
If the return of these macros are used only as boolean it makes sense to change the macros. It less intrusive.
I'm afraid though that most violations come from the construction `if (IS_ENABLED(CONFIG_BLA))`, that is a bigger problem, since this is always a constant branch and it triggers other violations too.