zephyrproject-rtos/zephyr #40538: mcuboot build fails with nrf52 internal RC oscillator

nobodyguy, 2021-11-21:
When I try to build my app with custom nrf52832 board (that has only internal RC oscillator configured as a LF clock source) it works. But when I try to build mcuboot with the same board, it fails.
Also mentioned here - https://devzone.nordicsemi.com/f/nordic-q-a/81418/bug-in-ncs-v1-7-1-with-32khz-rc-oscillator-and-mcuboot-enabled

Build of my app (I'm using T2 project structure):
```bash
user@6b89f0d24bcb:/workspaces/zephyr_workspace$ west build -b nrf52832_hangy -s hangy_firmware
-- west build: generating a build system
Including boilerplate (Zephyr base): /workspaces/zephyr_workspace/zephyr/cmake/app/boilerplate.cmake
-- Application: /workspaces/zephyr_workspace/hangy_firmware
-- Zephyr version: 2.7.0 (/workspaces/zephyr_workspace/zephyr), build: zephyr-v2.7.0
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52832_hangy
-- Cache files will be written to: /home/user/.cache/zephyr
-- Using toolchain: zephyr 0.13.2 (/opt/toolchains/zephyr-sdk-0.13.2)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /workspaces/zephyr_workspace/hangy_firmware/boards/arm/nrf52832_hangy/nrf52832_hangy.dts
-- Found devicetree overlay: /workspaces/zephyr_workspace/hangy_firmware/boards/nrf52832_hangy.overlay
-- Generated zephyr.dts: /workspaces/zephyr_workspace/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /workspaces/zephyr_workspace/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /workspaces/zephyr_workspace/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /workspaces/zephyr_workspace/build/zephyr/dts.cmake
Parsing /workspaces/zephyr_workspace/zephyr/Kconfig
Loaded configuration '/workspaces/zephyr_workspace/hangy_firmware/boards/arm/nrf52832_hangy/nrf52832_hangy_defconfig'
Merged configuration '/workspaces/zephyr_workspace/hangy_firmware/prj.conf'
Configuration saved to '/workspaces/zephyr_workspace/build/zephyr/.config'
Kconfig header saved to '/workspaces/zephyr_workspace/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/zephyr_workspace/build
-- west build: building application
[1/276] Preparing syscall dependency handling

[269/276] Linking C executable zephyr/zephyr_prebuilt.elf

[276/276] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      182568 B       200 KB     89.14%
            SRAM:       33956 B        64 KB     51.81%
        IDT_LIST:          0 GB         2 KB      0.00%
```

Build of mcuboot:
```bash
user@6b89f0d24bcb:/workspaces/zephyr_workspace$ west build -b nrf52832_hangy -d build/nrf52832_hangy_bootloader bootloader/mcuboot/boot/zephyr -- -DBOARD_ROOT=/workspaces/zephyr_workspace/hangy_firmware
-- west build: generating a build system
Including boilerplate (Zephyr base): /workspaces/zephyr_workspace/zephyr/cmake/app/boilerplate.cmake
-- Application: /workspaces/zephyr_workspace/bootloader/mcuboot/boot/zephyr
-- Zephyr version: 2.7.0 (/workspaces/zephyr_workspace/zephyr), build: zephyr-v2.7.0
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52832_hangy
-- Cache files will be written to: /home/user/.cache/zephyr
-- Using toolchain: zephyr 0.13.2 (/opt/toolchains/zephyr-sdk-0.13.2)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /workspaces/zephyr_workspace/hangy_firmware/boards/arm/nrf52832_hangy/nrf52832_hangy.dts
-- Found devicetree overlay: /workspaces/zephyr_workspace/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/dts.cmake
Parsing /workspaces/zephyr_workspace/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/workspaces/zephyr_workspace/hangy_firmware/boards/arm/nrf52832_hangy/nrf52832_hangy_defconfig'
Merged configuration '/workspaces/zephyr_workspace/bootloader/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/.config'
Kconfig header saved to '/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/autoconf.h'
[... 18 lines cut ...]
/opt/toolchains/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_mutex_unlock':
/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/syscalls/kernel.h:836: undefined reference to `z_impl_k_mutex_unlock'
/opt/toolchains/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_sem_init':
/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/syscalls/kernel.h:918: undefined reference to `z_impl_k_sem_init'
/opt/toolchains/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_mutex_init':
/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/syscalls/kernel.h:802: undefined reference to `z_impl_k_mutex_init'
/opt/toolchains/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a(temp_nrf5.c.obj): in function `k_sem_give':
/workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader/zephyr/include/generated/syscalls/kernel.h:953: undefined reference to `z_impl_k_sem_give'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /workspaces/zephyr_workspace/build/nrf52832_hangy_bootloader
```
Possible workaround is to use `CONFIG_CLOCK_CONTROL_NRF_K32SRC_SYNTH=y` option.
My custom board config is basically the nrf52832_mdk with these added options:
```
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=y
```

**Environment**
 * OS: Linux
 * zephyr-sdk-0.13.2
 * zephyr 2.7.0, but the same happens with the latest main and also with 2.6.0

Have anyone experienced this behavior? Or am I doing something wrong?

nobodyguy, 2021-11-22:
Disabling RC calibration with `CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n` also works (it skips temp_nrf5.c driver), but I wouldn't use this option.
Not sure why the temp_nrf5.c driver doesn't see the kernel functions.

carlescufi, 2021-11-24:
@tejlmand this feels like a build system problem, could you please take a look?

optimiseddesign, 2021-11-24:
I also experienced this, see Nordic ticket too;
https://devzone.nordicsemi.com/f/nordic-q-a/81418/bug-in-ncs-v1-7-1-with-32khz-rc-oscillator-and-mcuboot-enabled

> I would think one of the configs CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n or CONFIG_MULTITHREADING=y would be sufficient here

I ended up solving with CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=n to avoid the size overhead of multithreading which would enlarge the MCUBoot partition. @nobodyguy what issues could using this cause?

Note: this only needs to be in the MCUboot image (i.e. in \child_image\mcuboot.conf not the main project prj.conf).

nobodyguy, 2021-11-24:
https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/Kconfig#L685
This is probably the problem, it disables the multithreading support in mcuboot for nrf soc family.

@optimiseddesign Actually I'm not sure, I just quoted the Zephyr docs. It could be ok for mcuboot to keep the calibration disabled. Someone more skillful could give us an explanation of this.

nobodyguy, 2021-11-25:
Summary of what we (me, @optimiseddesign and Nordic's support) have found so far:

> Calibration routine for the 32kHz clock was changed (in nRFConnect SDk v1.6.0 I think). Basically, the 32kHz requires a calibration routine that depends on multithreading. So you will need to enable multithreading or disable the calibration in the sub-image that's not compiling. The temp_nrf5 files doesn't seem to handle calibration, but they do require multithreading, so please try enabling that in your application.

It seems that the root cause is that the calibration now requires a multithreading support, but the dependency between the **CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION** and **CONFIG_MULTITHREADING** config options is not defined in the code.
The build of mcuboot fails on **temp_nrf5** driver because the clock calibration code needs to sample temperature from nrf's internal temperature sensor.
It works in app builds because **CONFIG_MULTITHREADING** is turned on by default and it doesn't work in mcuboot builds because mcuboot's config explicitly disables it.

tejlmand, 2021-11-29:
@nobodyguy thanks for the thorough description here and on devzone.

I would like to point out there is one more option, and that is to set [`CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_MAX_SKIP=0`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_MAX_SKIP.html)

that will perform periodic calibration but not forcing `CONFIG_TEMP_NRF5`.

That means you can have:
```
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC_CALIBRATION=y
CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_MAX_SKIP=0
```

nobodyguy, 2021-11-29:
@tejlmand 
That's great, thank you very much :rocket:

carlescufi, 2021-11-30:
@nobodyguy is this resolved now? can we close the issue?

nobodyguy, 2021-11-30:
@carlescufi I think there is still a missing dependency between `CONFIG_TEMP_NRF5` and `CONFIG_MULTITHREADING`

tejlmand, 2021-12-01:
> @carlescufi I think there is still a missing dependency between `CONFIG_TEMP_NRF5` and `CONFIG_MULTITHREADING`

Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/40809

nobodyguy, 2021-12-01:
> > @carlescufi I think there is still a missing dependency between `CONFIG_TEMP_NRF5` and `CONFIG_MULTITHREADING`
> 
> Fixed here: #40809

Nice, thank you very much!

PR #40809 (2021-12-01): drivers: nrf5: kconfig: add missing dependency for TEMP_NRF5
Fixes: #40538

The TEMP_NRF5 setting requires MULTITHREADING, therefore add this
dependency to Kconfig.

The TEMP_NRF5 supports an alternative driver implementation using the
TEMP_NRF5_FORCE_ALT setting.
Because it cannot be known whether an alternative implementation has
the same dependency, then the MULTITHREADING is OR'ed with
TEMP_NRF5_FORCE_ALT, as to allow enabling of TEMP_NRF5 if an alternative
driver is provided.
commit 0c662409 (2021-12-01): drivers: nrf5: kconfig: add missing dependency for TEMP_NRF5

Fixes: #40538

The TEMP_NRF5 setting requires MULTITHREADING, therefore add this
dependency to Kconfig.

The TEMP_NRF5 supports an alternative driver implementation using the
TEMP_NRF5_FORCE_ALT setting.
Because it cannot be known whether an alternative implementation has
the same dependency, then the MULTITHREADING is OR'ed with
TEMP_NRF5_FORCE_ALT, as to allow enabling of TEMP_NRF5 if an alternative
driver is provided.
closed 2021-12-01 by carlescufi