zephyrproject-rtos/zephyr #40553: I2C write and read to 16 bit register address.

axdniki, 2021-11-22:
There are few chips which has 16 bit I2c register address. Current I2c library supports on only 8 bit address for start address i2c_burst_read & i2c_burst_write function.

aasinclair, 2023-03-08:
I have looked into this, and it has been discussed at the architecture working group.
The functionality was previously included in Zephyr, but was removed as part of #12588 (lots of details about the decision in this PR).

Unfortunately there is no way to implement this functionality in a portable manner, as i2c_transfer does not work correctly on all devices.

For portable write functionality, the caller should pack the address and register into an contiguous buffer and use i2c_write.

aasinclair, 2024-02-09:
This can't be done portably unfortunately (https://github.com/zephyrproject-rtos/zephyr/issues/20154).

closed 2024-02-09 by aasinclair