zephyrproject-rtos/zephyr #40560: Callbacks lack context information...

jimvert, 2021-11-22:
All callbacks (for the most part), lack the ability to have associated "user data" or context information.  This hampers development in C++ where context information is critical to get back to object instances.  Including user data among callback parameters as part of callback registration is a must.  At a minimum, a void * is necessary.

MikeDK, 2022-09-08:
Thanks @alwa-nordic for that workaround! This way the already existing user_data fields can be used for other useful things.
I created a small template class using your technique and wanted to share it, maybe it's useful for someone ;)
```
template <typename kObjectType, typename uObjectType>
class ObjectGlue
{
public:
    struct Glue
    {
        kObjectType kObject;
        uObjectType *uObject;
    };

    ObjectGlue(const kObjectType &kInitializer, uObjectType *uObject) : mGlue({kInitializer, uObject})
    {
    }

    virtual ~ObjectGlue()
    {
    }

    kObjectType *getObjPtrK()
    {
        return &mGlue.kObject;
    }

    static uObjectType *getObjPtrU(kObjectType *kObject)
    {
        Glue *glue = CONTAINER_OF(kObject, Glue, kObject);
        return glue->uObject;
    }

    Glue mGlue;
};
```

One could use it for example like this:

```
class myGluedClass
{
public:
   void Callback() { }
};

using WorkGlue = ObjectGlue<k_work, myGluedClass>;

void workDistributor(struct k_work *kWorkPtr)
{
    myGluedClass *glued = WorkGlue::getObjPtrU(kWorkPtr);
    glued->Callback();
}

void main()
{
   myGluedClass myObj;
   WorkGlue glue(Z_WORK_INITIALIZER(workDistributor), &myObj);

   // now submit some work, the C function gets called and this will then call the object's callback function
   k_work_submit(glue.getObjPtrK());
}
```
maybe the example does not make much sense, but i hope you get the idea ;)

henrikbrixandersen, 2022-11-02:
@jimvert Given the examples on how to obtain a context as presented in this issue - can this issue be closed?

jimvert, 2022-11-03:
As was mentioned by the first respondent, the presented method is a workaround.  As such, it doesn't address the core issue. 

In general, the lack of a "context" parameter is part of a broader API design issue which has to do with parameter lifetime.  The documentation states that bt_gatt_write_params, for example, must remain valid for callback invocation.  This means that both the bt_gatt_write_params structure and the "real" proposed C++ object context require separate memory management on behalf of the application developer.

One could effectively argue that few, if any elements of bt_gatt_write_params are interesting for callback execution, especially if one carries a separate "context" parameter into that callback.  As such, something like bt_gatt_write, should copy the elements of bt_gatt_write_params necessary for the write operation.  That would leave only the lifetime of the "context" parameter target to be managed by the application developer.

Again, this issue is more about API design decisions.  I'm sure there is good rationale around those decisions, but they simply aren't clear from the documentation.  Given the amount of experience garnered over the years that culminated into the Zephyr project, I would have expected closer scrutiny over those APIs...

alwa-nordic, 2022-11-03:
The lifetime requirement of parameters is a separate discussion. I think you have a valid point about the lack of user guides on memory management in Zephyr. You are welcome to create a new issue or RFC for that and I'll be happy to discuss it.

jimvert, 2022-11-04:
I've opened https://github.com/zephyrproject-rtos/zephyr/issues/51960 to address the issue of memory management in terms of the context parameter.

closed 2022-11-04 by jimvert