zephyrproject-rtos/zephyr #40561: BLE notification and indication callback data are difficult to pass to other threads...

jimvert, 2021-11-22:
BLE notifications and indication callbacks return only a data pointer and data length.  If these data are destined for a different thread, one must stand on their head to get the data to that thread without the underlying buffer being reclaimed.  Seeing as how a net_buf underlies notification and indication callback parameters, I don't understand why the net_buf isn't provided directly to notification and indication callbacks.  Handing the net_buf off to another thread is then a simple matter of incrementing the reference count in the callback and forwarding the net_buf off to the other thread, allowing the callback to then simply return.

alwa-nordic, 2021-11-24:
The current API forces the application to make a copy of relevant buffer contents before returning from callbacks. I agree that it would be nice to have a zero-copy mechanism for moving data into the application. But the net_bufs are owned by the HCI driver, and I worry that the application developer can too easily stall/deadlock the HCI driver if it runs out of net_bufs. This is a good opportunity to discuss if it would be feasible and desirable to make a API where the net_bufs are supplied by the application.

jimvert, 2021-11-24:
Even if it's a user-supplied buffer, the chance of stalling the HCI driver is still a possibility in the zero-copy scenario unless it's the user who provides the indication response instead of it being automatically performed by the stack.  Notifications end up being trickier, no?  If the user hasn't provided a buffer for the notification and the driver is ready to deliver it, what happens?  

It doesn't seem that putting the buffer in the user's hands makes this problem easier.  Also, it's just another set of buffer allocations with which to contend.  Let the buffers be managed by the HCI driver.  Another option might be making indication responses explicit -- the indication data are not released until an explicit "send indication response" is issued from the user.

jori-nordic, 2023-02-16:
This seems like adding even more complexity to the stack for questionable benefits.
Seeing that it's been more than a year without any PRs or prototypes, it seems this is not really critical functionality.

Closing as won't do. @jimvert if you have a proof of concept you'd like us to review, feel free to push a PR and re-open this issue.

closed 2023-02-16 by jori-nordic