zephyrproject-rtos/zephyr #40562: net_buf and simple_net_buf need to be combined into a single type...

jimvert, 2021-11-22:
net_buf and simple_net_buf should be abstracted into a single type instead of have two types.  The differences should be abstracted away from the developer.  Right now, it's difficult to maintain a consistent application-level API having two different types.

carlescufi, 2023-11-22:
@jhedberg assigned to you, any thoughts on the matter?

jhedberg, 2023-11-22:
I don't see how they could be abstracted to a single type, since they don't implement the same APIs (net_buf is a superset of net_buf_simple), i.e. if you have a net_buf_simple, you can't pass it to APIs that expect a full net_buf. For the other direction, you can get a net_buf_simple out of a net_buf with: `struct net_buf_simple *simple = &buf->b`.

closed 2024-12-16 by jhedberg