zephyrproject-rtos/zephyr #40563: Polling API lacks support for timers and mailboxes...

jimvert, 2021-11-22:
More thought needs to be put into the polling API.  As an example, the polling interface lacks direct support for timers and mailboxes.  One has to stand on their head to fit these constructs into the polling abstraction.

henrikbrixandersen, 2022-11-02:
Please provide more context and examples as to the need for standing on your head.

jimvert, 2022-11-03:
I'm unsure how to respond to this follow-up.  As I mention in my initial comment, "the polling interface lacks direct support for timers and mailboxes."  This means that one can't _directly_ use k_poll to determine the expiration of a timer, for example.  There are likely many workarounds (i.e. standing on one's head), but one way is to have timer expiration trigger a signal event (which can be polled).  

While there may be valid reasons to have excluded timers and mailboxes from being k_poll'ed, those reasons aren't clear and from a high level seem to have been a design oversight.