zephyrproject-rtos/zephyr #40569: bisected: kernel.common.stack_protection_arm_fpu_sharing fails on mps3_an547

carlocaione, 2021-11-22:
`kernel.common.stack_protection_arm_fpu_sharing` is failing for `mps3_an547`

```
*** Booting Zephyr OS build v2.7.99-1639-g73a957e4b316  ***
Running test suite fatal
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
```
Result of bisection:
```
fcad2b37e39196ec6cc2baa233360b54a499fc1e is the first bad commit
commit fcad2b37e39196ec6cc2baa233360b54a499fc1e
Author: Ying ming <mingx.ying@intel.com>                                                                  
Date:   Tue Mar 9 16:37:57 2021 +0800
                                                     
    test: fatal exception: Add code description and testcases
                           
    Improve the test case of CPU exception.
    Add equivalence classes and input partition testing
    when give an integer reason code.                                                                     
                                                                                                          
    Signed-off-by: Ying ming <mingx.ying@intel.com>                                                       
                                                     
 tests/kernel/fatal/exception/src/main.c | 62 +++++++++++++++++++++++++++++++++                                                                                                                                      
 1 file changed, 62 insertions(+)                                                        
```

carlescufi, 2021-11-25:
When this test was introduced, it worked on all? (most) boards. Then the board in question was added and this test fails on it. So it could be an issue with the test, an issue with the board or an issue with the actual code it's testing.

carlocaione, 2021-11-25:
Yeah, I investigated a bit and it seems that the actual problem was introduced by https://github.com/zephyrproject-rtos/zephyr/commit/70dd3d6b0deec17eab6e0888d458b7e5aa1481da

@stephanosio can you take a look?

stephanosio, 2021-11-25:
> @stephanosio can you take a look?

@carlocaione Sure.

theotherjimmy, 2021-12-07:
This might be the same root cause as the issue I just created, #40970, which is a different test failure on the same platform because qemu was missing MVE for this platform.

yingmingx, 2021-12-08:
Cancel my assignment due to conversion to another team.

galak, 2021-12-08:
https://github.com/zephyrproject-rtos/zephyr/pull/41018

theotherjimmy, 2022-01-13:
Confirmed to be MVE, it's masked by 91d4b7766ceee7fabb5caa026ba181c47286e28b, which disables MVE for qemu, which is used for testing.

closed 2022-01-13 by carlocaione